BehaviorTree/BehaviorTree.CPP #2: Conan Package Support

uilianries, 2018-10-18:
Hello,

Do you know about [Conan](https://conan.io)?

Conan is modern dependency and package manager for C/C++ and would be great if your library would be available via package manager for other developers.

All Conan packages are available on [Conan Center](https://bintray.com/conan/conan-center), where is possible search, install and upload any official Conan package.

As your project uses zmq, we already have such dependency solved by Conan. So you are able to run BehaviorTree with zmq pre-built.

We could support you writing a recipe and publishing your packages on Bintray.

If you have any questions, just ask :-)

Regards!

facontidavide, 2018-10-19:
I looked in the conan center and I was not able to find "ZMQ". Furthermore, the command

conan search "*z*" -r=conan-center

Doesn't return anything relevant.

uilianries, 2018-10-19:
Sorry, ZMQ is available by Bincrafters team:

https://bintray.com/bincrafters/public-conan/zmq%3Abincrafters

Bincrafters is a team for OSS developers to collaborate on binary packaging of libraries. We have helped Conan and the community.

ZMQ is not official on Conan center yet, but is close to be promoted.

facontidavide, 2018-10-19:
summarizing, what should I do to use ZMQ from Bincrafters?
I am 100% new to Conan.

Eventually, my build MUST support:

- Compile without CONAN
- Compile with Catkin (ROS)
- Vanilla compiling (if neither Catkin nor Conan is detected).

If you want to contribute with a PR, you are welcome

uilianries, 2018-11-14:
Hi! I've started the recipe for Behavior Tree and I have two questions:

- Who is the author of this project? 
  I found 3 main members in collaborators, but I don't have sure if this project is there is only one father.

- What is the correct name for this project?
  BehaviorTree.CPP is the name of this project, but in CMakeLists.txt the name is   `behavior_tree_core`

Regards!

facontidavide, 2018-11-14:
The author is mainly me, or at least the main maintainer. Davide Faconti.
The "human readable" name of the project is BehaviorTree.CPP, but I don't know if it is ok for a package manager.

In particular, ROS accepts only projects in snake case.

I might rename  __behavior_tree_core__ as  __behaviortree_cpp__, to be more consistent.

uilianries, 2018-11-14:
Thanks for you quick answer!

uilianries, 2018-11-14:
I have few questions again:

- Is there any idea supporting shared library?
  I found `add_library(behavior_tree_core STATIC ${BT_Source} )`. The library is forced to be static ever.

- Could fPIC be more relaxed?
  I found `target_compile_options(behavior_tree_core PRIVATE "-fPIC")`. However cmake provides `POSITION_INDEPENDENT_CODE` which is possible to be used as fPIC but as optional.

- Is there any idea supporting install target?
  I didn't find `intall` section in your cmake file. It could help all customers.

- Do you want export BehaviorTree using cmake config file?
  CMake provides helpers to exports all paths and libraries names in a single file.

- Is BehaviorTree supported on Windows and Macos?
  I found `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -Werror=return-type -Wall -Wpedantic -Wattributes")` that is related to Linux (clang, gcc)

As I have some experience with CMake, I would be happy helping this project updating your cmake as well. I could create a separated PR introducing all changes.

facontidavide, 2018-12-05:
Done

closed 2018-12-05 by facontidavide