BehaviorTree/BehaviorTree.CPP #3: Building with colcon for ROS2

mjeronimo, 2018-10-22:
Can you integrate support for colcon in the CMakeLists.txt? Here's what I have:

https://github.com/mjeronimo/BehaviorTree.CPP/commit/c489ce52dbdf81b5d4c3ffadd9605f49ffd3ab12

facontidavide, 2018-10-23:
You probably noticed that I carefully created a CMakeLists.txtx file that works both with and without catkin ;)

This means that even the addition of colcon need to be done in a similar way, to be sure that it "automagically" work no matter if you use ROS1, ROS2 or "vanilla cmake".

I will take a look at this

mjeronimo, 2018-10-23:
Yes, I figured you'd support all of them. Thanks.

mjeronimo, 2018-11-29:
@facontidavide Any updates on this one? I may be able to take a look at this one if you haven't already.

facontidavide, 2018-12-05:
Fixed by @mjeronimo  in PR #27

closed 2018-12-05 by facontidavide