BehaviorTree/BehaviorTree.CPP #4: TinyXML symbols exported from behavior tree library

mjeronimo, 2018-10-22:
With the currently build scripts, the tinyxml2 symbols are exported from the behavior_tree_core shared object. The ROS2 middleware also embeds the tinyxml2 library and exports the symbols as well. The ROS2 build will use the tinyxml2 library installed on the target platform, while BehaviorTree.CPP will use its own version (the tinyxml2 source is with the BT library code). If these libraries are different versions only one tinyxml2 version will be found when link a ROS2 executable. Therefore, one the ROS2 middleware or BT library will be using the wrong version. 

TinyXML2 supports the gcc's visibility attribute. I made the following change to work around this:

https://github.com/mjeronimo/BehaviorTree.CPP/commit/f059c2035eda28a9688bf063a428fa2d44cfa9e7

With this, by default, the tinyxml2 symbols will not be exported and will therefore, not conflict.

mjeronimo, 2018-10-22:
Also, once this change is made, it introduces a compiler warning:

```
‘BT::XMLParser’ declared with greater visibility than the type of its field ‘BT::XMLParser::doc_’ [-Wattributes]
 class XMLParser
```

This is because the BT:XMLParser symbol has default visibility, while one of its members, the doc_ field:

```
class XMLParser {
...
tinyxml2::XMLDocument doc_;
}

```
has hidden visibility (since it is a tinyXML2 type). I've avoided this warning, by using the following:

```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wattributes"
#include "behavior_tree_core/xml_parsing.h"
#pragma GCC diagnostic pop
```

However, perhaps you could have your own type to wrap the tinyxml type and completely encapsulate the use of tinyxml2.

facontidavide, 2018-10-23:
Actually, I do not want to export/expose at all the implementation of tinyXML2 to the user. i will see how to change this...

facontidavide, 2018-10-23:
I just pushed a commit that should solve the problem. Let me know what you think.

c119eaf89536acb34d0be575e53193442cfd0599

I used the Pimpl idiom to completely hide tinixml2 from the eye of the user of the library.

mjeronimo, 2018-10-23:
I'm seeing the following warning when compiling BehaviorTree.CPP:

```
/home/mdjeroni/src/navstack_dependencies/src/BehaviorTree.CPP/src/xml_parsing.cpp:21:19: warning: ‘BT::XMLParser::Pimpl’ declared with greater visibility than the type of its field ‘BT::XMLParser::Pimpl::doc’ [-Wattributes]
 struct XMLParser::Pimpl
```
However, no issues when I build and link against BT.CPP.

facontidavide, 2018-10-24:
Fixed the compilation warning

closed 2018-10-24 by facontidavide