BehaviorTree/BehaviorTree.CPP #5: Recommended behavior tree editor?

mjeronimo, 2018-10-25:
I see on your site that you have a screenshot of a Behavior Tree editor. Do you happen to have a recommendation of which to use? I'd like to output XML that can be used w/ the BehaviorTree.CPP library to automatically construct the tree. Also, of course, I'd have to be able to add custom nodes to the editor so that they can be placed in the tree and parameters for each action specified.

facontidavide, 2018-10-25:
The recommended editor is Groot https://github.com/BehaviorTree/Groot

The only "problem" that makes it a pre-alpha is the somehow cumbersome way to create new custom Actions ( I will fix it in November).

Right now custom TreeNodes can be added to the palette of Groot only through XML file. 

Furthermore, the GUI is not meant to contain the executor as well. They should remain two different processes that communicate in real-time thought TCP.

I have to write some documentation to show how this work. I hope to have it by the end of Novemeber.

Cheers

facontidavide, 2018-10-25:
To get an idea of how Groot works, you can take a look at the Vimeo channel https://vimeo.com/manage/albums/5201743

mjeronimo, 2018-10-25:
Thanks for the info and the pointer to the video. Can you add me to the permissions to that Vimeo folder so that I can view the video?

facontidavide, 2018-10-25:
edited the link, my bad

facontidavide, 2018-11-15:
The readme clearly points out to Groot. I will close this

closed 2018-11-15 by facontidavide