BehaviorTree/BehaviorTree.CPP #6: Compile-time error reference M_PI when building on Ubuntu 18.04

mjeronimo, 2018-10-25:
examples/t04_blackboard.cpp uses the macro M_PI, which is not defined, resulting in:

```
src/behavior_tree_core/examples/t04_blackboard.cpp:44:34: error: ‘M_PI’ was not declared in this scope
     const Pose2D mygoal = {1, 2, M_PI};
                                  ^~~~
```
Perhaps add the following: 

```
#ifndef M_PI
#define M_PI <value>
#endif
```

facontidavide, 2018-10-29:
fixed

closed 2018-10-29 by facontidavide

mjeronimo, 2018-11-21:
@facontidavide  I just pulled the latest code and this doesn't seem to be fixed:

```
src/BehaviorTree.CPP/examples/t04_blackboard.cpp: In function ‘BT::NodeStatus CalculateGoalPose(BT::TreeNode&)’:
src/BehaviorTree.CPP/examples/t04_blackboard.cpp:44:34: error: ‘M_PI’ was not declared in this scope
     const Pose2D mygoal = {1, 2, M_PI};
```

facontidavide, 2018-11-21:
I'll be damned... fixed...