BehaviorTree/BehaviorTree.CPP #7: Crash on BT load

StefanoSoffiaUTRC, 2018-11-02:
GRoot crashes when loading the following BT xml:

```
<root>
    <!--------------------------------------->
    <BehaviorTree ID="ReceiveGuest">
    </BehaviorTree>
    <!--------------------------------------->
    <TreeNodesModel>

        <SubTree ID="ReceiveGuest"/>

        <Control ID="Fallback"/>
        <Control ID="Sequence"/>
        <Control ID="SequenceStar"/>
    </TreeNodesModel>
    <!--------------------------------------->
</root>
```
It looks like it is trying to dereference a null pointer:

```
registered  "SequenceStar"
AddressSanitizer:DEADLYSIGNAL
=================================================================
==15032==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x5606027bf02b bp 0x7ffd2cc13ff0 sp 0x7ffd2cc13fe0 T0)                                                                                
==15032==The signal is caused by a READ memory access.
==15032==Hint: address points to the zero page.
```

facontidavide, 2018-11-02:
Thank you for reporting. I will fix it as soon as possible

facontidavide, 2018-11-02:
Hi.

I just realized that you mentioned Groot. This problem is not related to the core library, anyway I added a unit test to show that the error in the XML is reported.

A <BehaviorTree> must not be empty.

I moved this issue to Groot: https://github.com/BehaviorTree/Groot/issues/3

closed 2018-11-02 by facontidavide