BehaviorTree/BehaviorTree.CPP #10: Better CMake Support

uilianries, 2018-11-15:
Hi!

I have few questions again:

- Is there any idea supporting shared library?
  I found add_library(behavior_tree_core STATIC ${BT_Source} ). The library is forced to be static ever.

- Could fPIC be more relaxed?
  I found target_compile_options(behavior_tree_core PRIVATE "-fPIC"). However cmake provides POSITION_INDEPENDENT_CODE which is possible to be used as fPIC but as optional.

- Is there any idea supporting install target?
  I didn't find intall section in your cmake file. It could help all customers.

- Do you want export BehaviorTree using cmake config file?
  CMake provides helpers to exports all paths and libraries names in a single file.
- Is BehaviorTree supported on Windows and Macos?
  I found set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -Werror=return-type -Wall -Wpedantic -Wattributes") that is related to Linux (clang, gcc)

As I have some experience with CMake, I would be happy helping this project updating your cmake as well. I could create a separated PR introducing all changes.

facontidavide, 2018-11-15:
Hi,

let me answer in order:

1) I Guess that there is no problem to remove STATIC

2) I need to check that the generate plugins would still work as expected

3) Yes, please check the open pull request. #9 

4) I am not familiar with it, any help is welcome.

5) I want to make it available for Windows too, I am just too busy to focus much on it. With a little work, it should be easy.

facontidavide, 2018-11-15:
Please take a look at this branch... https://github.com/BehaviorTree/BehaviorTree.CPP/tree/cmake_fix
I have to think more about the installation.

facontidavide, 2018-11-16:
I just release changes that should solve most of the issues you mentioned, except number 4, where I need someone to send me some references about packaging with CMake.

closed 2018-11-16 by facontidavide

uilianries, 2018-11-16:
Thanks for your answer! I will provide a PR with the number 4 (cmake export).