BehaviorTree/BehaviorTree.CPP #12: set priority of sequence node's children

liuxin00738, 2018-11-16:
Hi, I am planning to use the behavior tree, and I find the great work you have done. I have one question about dynamically changing the priority of the child of a sequence node. Is this possible to do in this framework? 

In the readme I saw that "It allows the creation of trees at run-time, using a textual representation (XML)." I did not see where in the tutorial to do this. If we cant set the priority of the child, maybe I can use this do realize it. 
Thanks in advance!

Best regards,
Xin Liu

facontidavide, 2018-11-16:
The Sequence is created at run-time, but once it is created, the order is not changed.
In other words, the XML is loaded only at the beginning.

NEVERTHELESS, we can work together to create a special Sequence where the order is changed dynamically.
How would you do that? How do you want to change the priority at run-time?

Let's figure this out together.

liuxin00738, 2018-11-16:
Sorry I think I made a mistake. I mean fallback node.
So say normally for a fallback node, I would try method A, then method B,
then method C.
However, if I have from vision algorithm suggest me method B would be more
likely to success this time, it would be more beneficial to change the
sequence. This could make the system success faster.

I just start yesterday to read this method so my thinking could be totally
not to the point, but I was thinking this:
Either (1) for the fallback node, we use a priority queue so the parent
will choose the one with higher priority, and expose a SetProbability in
the sub-tree to change its sequence. Set it to zero and this subtree will
never be executed.
(2) we allow dynamic insert/attach a subtree into the parent tree, by
somehow pushing an array of subtree pointers.
(3) we write to xml file and reload, although this seems not to the best
for speed.

I feel the first is kind of plausible, any thoughts?

facontidavide, 2018-11-18:
2) and 3) are against what I consider good design practices ;)
Trees shall not be modified at run-time. 

This means that we should focus on 1) 
I suggest having a special ControlNode that execute its children based on a priority value.
The question remains: how do we set such value?
Since you want to change it at run-time, it MUST be a Key/Value in the Blackboard.

Brainstorm:

Supposing this PriorityNode has 4 children, if the Blackboard contains the string "2;1;3;-1"

- Second child is executed first;
- First child is executed second;
- Third child is executed third.
- Fourth child is not executed.

The next question is: if any child fail should we stop or continue?

What do you think?

liuxin00738, 2018-11-18:
I agree the blackboard is better. If any child fail we should continue until there is no child that has a char that is positive. We could give default value of 1 for all children so no change in behavior if people dont want to use this feature. 

The application I have in mind is that these children will call some time-consuming action nodes,  so if we choose the child that is more likely to success, the speed will be faster.

facontidavide, 2018-11-19:
I have created a new node in this branch:

https://github.com/BehaviorTree/BehaviorTree.CPP/tree/priority_fallback

Please take a look, use it and tell me if it works for you.

Cheers

JulianKu, 2021-12-20:
I just stumbled upon this issue thread while looking for a way to combine Machine Learning (Hierarchical Reinforcement Learning to be precise) with Behavior Trees.
@facontidavide I guess the Priority Fallback Node that you came up with in the branch you mention could be exactly what I am looking for. I saw that it never made it to main branch and the the `priority_fallback`-branch got deleted.
Any chance to revive it?

I'd like to use a Neural Network in this Priority Fallback Node which outputs values ([0, 1]) for each child node so that the node with highest value gets tried first; upon failure of one child, the one with next highest value could be tried out (maybe down to a certain value threshold after which the entire Priority Fallback Node returns failure).