BehaviorTree/BehaviorTree.CPP #13: Parallel node not able to be registered and used in an XML description

mjeronimo, 2018-11-16:
To implement runing planning and control in parallel, I'd like to be able to use the ParallelNode in an XML description, something like this:
```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <SequenceStar name="root">
      <ComputePathToPose endpoints="${endpoints}" path="${path}"/>
      <Parallel threshold_M="1">
        <FollowPath path="${path}"/>
        <Sequence>
          <RateController hz="2">
            <ComputePathToPose endpoints="${endpoints}" path="${path}"/>
          </RateController>
          <UpdatePath/>
        </Sequence>
      </Parallel>
    </SequenceStar>
  </BehaviorTree>
</root>
```
However, because the ParallelNode has the following signature:

`BT::ParallelNode::ParallelNode(const std::string& name, int threshold_M)
`

it can't be registered and used in an XML description of the tree. I've worked around this for now by making the threshold value have a default value.

facontidavide, 2018-11-18:
I will look into this issue tomorrow. Thanks for reporting

facontidavide, 2018-11-19:
Hi, I just fixed it with commit 0abbd6098c74efd84e5eaa0221a535bb5adb6301

By the way, if you want to share the code of RateController, that would be great; I was considering to implement a similar Decorator Node

closed 2018-11-19 by facontidavide

mjeronimo, 2018-11-20:
Great, thanks! Here's what I did: https://github.com/mjeronimo/navigation2/blob/parallel-planning-and-control/nav2_tasks/include/nav2_tasks/rate_controller_node.hpp. It was a quick version. Perhaps you've got a better way to do it.

mjeronimo, 2018-11-21:
@facontidavide You've updated the ParalleNode class so that it can be registered (thanks!), but it is not actually registered yet in the BehaviorTreeFactory class. It seems that it should be registered there along with the other control nodes. I can register it myself, but just wanted to check if that's what you intended.

facontidavide, 2018-11-21:
I totally forgot it. You probably noticed that I don't use this node myself ;)

 It was written by @miccol  :P

Fixed.