BehaviorTree/BehaviorTree.CPP #15: question about integration legacy code with behavior tree

liuxin00738, 2018-11-20:
Hi Davide,

I currently have some legacy code which I want to use together with BehaviorTree, one of them is a class which provides some methods that I want to wrap into the action nodes. However, these member functions take some input parameters. 

From my understanding, if I did not use the inheritance, I could not use parameters, right? If I use inheritance, it seems there is only one action can be defined by each class. Is this correct?
Is there a way to define multiple actions for one class, as well as using node or blackboard parameters?

Best Regards,

Xin

facontidavide, 2018-11-20:
I added a new tutorial just for you ;)

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/examples/t06_wrap_legacy.cpp

I hope it helps. If you have any question, let me know.

closed 2018-11-20 by facontidavide

liuxin00738, 2018-11-20:
It works! Thanks!