BehaviorTree/BehaviorTree.CPP #16: Blackboard precondition decorator throws std::invalid_argument

Thordreck, 2018-11-22:
Hi,
It seems there's an issue with the `BlackboardPreconditionNode` decorator, or it may be a misunderstanding on my part on how it's supposed to work. Setting the "expected" field with the wildcard value "*" throws a `std::invalid_argument` from `std::stod`. I went ahead and check what's going on, and the problem seems to come from how the checks are arranged in the `tick()` function of the decorator:

```
        if (blackboard() &&                     //blackboard not null
            getParam("key", key) &&             // parameter key provided
            getParam("expected", expected) &&   // parameter expected provided
            blackboard()->get(key, value) &&    // value found in blackboard
            (value == expected ||
             initializationParameters().at("expected") == "*"))   // is expected value or "*"
        {
            return child_node_->executeTick();
        }
        else
        {
            return NodeStatus::FAILURE;
        }
```

The `getParam("expected", expected)` gets called even if the value is "*" and it ends up calling `convertFromString<double>("*")` in `tree_node.h`.
Rearranging the conditions a bit fixes this. Something like this:

```
        if (blackboard() && getParam("key", key) && blackboard()->get(key, value))
        {
            if (initializationParameters().at("expected") == "*" || (getParam("expected", expected) && value == expected))
            {
                return child_node_->executeTick();
            }
        }
        return NodeStatus::FAILURE;
```

facontidavide, 2018-11-22:
Thanks for reporting, I will check it out

facontidavide, 2018-11-22:
fixed

facontidavide, 2018-11-22:
Do not update yet, I realized that the I just changed the semantic and I need to revert the change

facontidavide, 2018-11-22:
Ok, take a look now. I think it is fixed.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp/decorators/blackboard_precondition.h#L45-L70

facontidavide, 2018-11-23:
Let me know if it works as expencted

closed 2018-11-23 by facontidavide

Thordreck, 2018-11-23:
Hi, sorry for the delay. I tested it and it's working great. Thank you for the prompt reply.
Cheers.