BehaviorTree/BehaviorTree.CPP #17: Support inclusion or aggregation of other XML files

v-lopez, 2018-11-23:
The use case is simple, if I create a BehaviorTree that implements a funcionality in an xml, I'd like to be able to include it from multiple XML's without having to duplicate it.

facontidavide, 2018-11-23:
[image]

facontidavide, 2018-12-03:
Ok, I implemented the feature in a new branch. Once you give me some feedback I will merge it into master.

https://github.com/BehaviorTree/BehaviorTree.CPP/tree/include_xml

To see it in action, execute __bin/t07_include_trees__ passing as first argument either the absolute or relative path to  __examples/test_files/subtree_test.xml__.

Cheers

v-lopez, 2018-12-03:
Looks great. 

ROS users could probably implement on their own application, some sort of preprocessing of the xml to expand paths relative to a package, ie expand `<ros_include pkg="other_pkg" path="subtrees/Talk.xml" />` into `<include path="/absolute_path_to_other_pkg/subtrees/Talk.xml" />`

facontidavide, 2018-12-03:
Indeed. as usually, I prefer to focus on platform independent features first and ROS specific ones later.

facontidavide, 2018-12-03:
Please take a look at db72762748a34b20dc44fa23be1b7562ed469586

I didn't test it through yet, but it should work.

v-lopez, 2018-12-03:
I tested it modifying your example with:
```    <include ros_pkg="pal_statistics" path="foo.xml" />```

And it worked like a charm!

facontidavide, 2018-12-03:
Nice!!!

closed 2018-12-03 by facontidavide

fafux, 2021-02-05:
HI, I faced a problem in including xml files. I execute bin/t10_include_trees passing as first argument the path to examples/test_files/subtree_test.xml. Below the output:
```bash
terminate called after throwing an instance of 'BT::RuntimeError'
  what():  CheckTemperature is not a registered node, nor a Subtree
Aborted (core dumped)
```