BehaviorTree/BehaviorTree.CPP #22: RetryUntilSuccesful fail to get parameter from blackboard

liuxin00738, 2018-11-29:
Hi,
Recently I am trying to set a blackboard parameter, then ask retryuntilsuccessful to retry this num of times. However, it keeps core dumpt and says stol, invalid argument. 
This error is easy to reproduce with t05_crossdoor.cpp. Here is my version of code which will error out.
I tried to give a unsigned int, I am not sure what the type this num_attempts should be. 

[t05_crossdoor.cpp.tar.gz](https://github.com/BehaviorTree/BehaviorTree.CPP/files/2627773/t05_crossdoor.cpp.tar.gz)

facontidavide, 2018-11-29:
Ok, I see what is happening.

getParam is called in the constructor, but the blackboard is assigned only later.
I can fix this, but I am afraid that this error will happen a lot to other people writing their own nodes.
I have to think about a long term solution

facontidavide, 2018-11-29:
I just pushed another fix related to that. Please pull from master and let me know if anything looks weird.

liuxin00738, 2018-12-01:
I tried it today and it works. Thanks!