BehaviorTree/BehaviorTree.CPP #26: profiling tools request

liuxin00738, 2018-12-04:
Hi Davide,
Is there a way to profiling the tree, i.e., how much a step take? I know there is a jason log and I can parse that. I am just wondering if you have that already. Also, is there a way to make the time stamp accurate to micro second?

Best Regards,
Xin

facontidavide, 2018-12-04:
I guess your best shot is to look at the way loggers work. They subscribe to status changes. It is a simple observer pattern. If you give more details i may help you with more concrete suggestions

facontidavide, 2018-12-04:
I guess that you want to create a logger that give you some statistics about __leaf nodes__. Am I right? Or you want to collect time related to other nodes too? 
may be this may include min, max and average time, or even an histogram.
I think it is very easy to implement.

facontidavide, 2018-12-10:
If you give me more details about what you really want, we can get this done

liuxin00738, 2018-12-10:
Sorry for the late reply. I was not working on this front for the last week.

What I have in mind is this:
Say I have some actions, and I want to know, in a tick, how much time each action is called, and how long each action takes. This way I want to use it to pinpoint which action is called most of the time, and take most of the time. Then improve that action.

I think MinitraceLogger might be able to do this, but I did not look into it. I didn't need this function now so I didnt keep digging. But if you can add it, that would be great.

liuxin00738, 2019-01-15:
Finally I am back to here. I add a few lines to the bt_file_logger.cpp and save the log also into a txt format. This will slowdown the logging, but it works for my purpose. 

The main change is inside the callback() function of this file:
txtfile_<<timestamp.count()<<","<<node.name()<<","<<prev_status<<","<<status<<std::endl;
I am closing this issue.

closed 2019-01-15 by liuxin00738