BehaviorTree/BehaviorTree.CPP #28: Release to ROS2 build farm?

mjeronimo, 2018-12-05:
Could we a version of the library released to the ROS2 build farm? As I understand it, this requires rosdep keys and preparing a bloom release. This would allow us to reference the BT library in our official Navigation2 software for the Crystal release of ROS2.

facontidavide, 2018-12-05:
Sure. Which instruction should I follow to do it? These ones?
https://index.ros.org/doc/ros2/Releasing-a-ROS-2-package-with-bloom/

I have this installed:
      python-bloom  (0.6.9-1).
     python-catkin-pkg  (0.4.9-100).

Which apparently are sufficiently recent.

But when I run the bloom command, I get this error:

         bloom-release behaviortree_cpp --track crystal --rosdistro crystal
        
          > Unknown release: 'crystal'. Valid release names are: groovy, hydro, indigo, jade, kinetic, lunar, melodic

facontidavide, 2018-12-05:
https://github.com/ros2/rosdistro/pull/371

facontidavide, 2018-12-05:
merged

nuclearsandwich, 2018-12-05:
@facontidavide it doesn't look like the debian release information was pushed to the release repository https://github.com/BehaviorTree/behaviortree_cpp-release

Looking at the existing BehaviorTree.CPP source it appears to rely on catkin and roslib which aren't used by ROS 2.

@mjeronimo it looks like there is porting work that needs to be done to bring BehaviorTree.CPP to ROS 2.

facontidavide, 2018-12-05:
both roslib and catkin can be removed, since the library is meant to be compiled with "vanilla" cmake and doesn't require ROS at all.

If I just remove them from package.xml will it work? 
I guess the solution would be to push a different branch

facontidavide, 2018-12-05:
Please take a look at this: https://github.com/BehaviorTree/BehaviorTree.CPP/tree/ros2

Theoretically it should not complain now, but bloom_release complains:

```
Could not resolve rosdep key 'catkin'
Failed to resolve catkin on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'catkin', aborting.
catkin is depended on by these packages: ['behaviortree_cpp']
<== Failed
Could not resolve rosdep key 'roslib'
Failed to resolve roslib on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'roslib', aborting.
roslib is depended on by these packages: ['behaviortree_cpp']
```

I set __Upstream Devel Branch__ to __ros2__ but it still take the package.xml from master.

Anyone knows what I am doing wrong?

facontidavide, 2018-12-05:
I MIGHT have fixed it. If I didn't, I need someone more knowledgeable than me to help :/

https://github.com/ros2/rosdistro/pull/372

nuclearsandwich, 2018-12-05:
Ah, I missed the ros2 branch of BehaviorTree.CPP although I was quite certain I checked for one. An outside glance at the update suggests it matches with what a ROS 2 package would expect. I've merged the release and we'll see how it builds. Thanks for iterating @facontidavide!

facontidavide, 2018-12-05:
> I missed the ros2 branch 

You missed it because I created it 1 hour ago :dagger: 
Cheers

closed 2018-12-05 by facontidavide

mjeronimo, 2018-12-05:
@facontidavide  Thanks!

nuclearsandwich, 2018-12-05:
> You missed it because I created it 1 hour ago ;)

I am really impressed by this. :fireworks:

nuclearsandwich, 2018-12-06:
@mjeronimo are you able to use the `ros2` branch from source in your work downstream? Now that the [development job is passing](http://build.ros2.org/view/Cdev/job/Cdev__behaviortree_cpp__ubuntu_bionic_amd64/4/) I was going to suggest another release, but it's not a bad idea to field test it before doing so if you have the opportunity.

mjeronimo, 2018-12-06:
@facontidavide @nuclearsandwich Yes, just testing it and the ros2 branch works fine. I'll update our repo to use that tag.

nuclearsandwich, 2018-12-06:
In that case at @facontidavide a new release for Crystal with the latest changes would be very welcome.

facontidavide, 2018-12-06:
OK