BehaviorTree/BehaviorTree.CPP #33: Accessing the blackboard from derived classes

mjeronimo, 2018-12-10:
I'm using generation of the BT from an XML document. Also using the blackboard to share some data between nodes. When BT nodes are created with buildTreeFromText, the blackboard is not yet available to nodes in their constructor (since it is not set yet on the node). Therefore, in the tick() override, I have to check if this is the first time tick() has been called so I can get/set data intializing data from the blackboard after construction. Perhaps the library itself should have an onInit() function, in addition to tick() and halt(). The derived classes could override this method and know that the blackboard is available.

facontidavide, 2018-12-10:
I am kind of aware of this problem. 

To see how I deal with this problem take a look to the "design pattern" I used.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/decorators/retry_node.cpp#L28-L51

As you know, it is not possible to access the blackboard in the constructor. I am afraid there is a little I can do about it.

On the other hand, blackboard should be available at the first tick().

When a parameter points to a blackboard (pattern ${KEY}) the way to avoid problems is to use the method __isBlackboardPattern__, store in a boolean the result and:

- if false, it is safe to read the parameter in the constructor.
- if true, the parameter should be read at the beginning of __each__ tick.

> Therefore, in the tick() override, I have to check if this is the first time tick() has been called so I can get/set data intializing data from the blackboard after construction

I have the feeling that reading the blackboard only once is potentially an anti-pattern (and in some cases an actual error).

In fact, we don't know if the value inside the blackboard will change between a tick and the following one.
For this reason, even if it seems like a waste of respurces, the only right thing to do is to __get__ the value from the blackboard at each tick.

About the __onInit()__ method, I have no strong feelings about it, but when do you want to call it? Only once after construction, just before the very first tick()?

facontidavide, 2018-12-10:
>  When BT nodes are created with buildTreeFromText, the blackboard is not yet available to nodes in their constructor

To be precise, this is not really a limitation of buildTreeFromText; the only way to initialize the blackboard in the constructor would be to pass it as an argument.

mjeronimo, 2018-12-10:
Interesting about the anti-pattern. There seems to me to be two classes of data here. First, those items that might logically be part of a constructor and those that are dynamically updated (and shared) during the execution of the BT. For the first kind, there isn't an opportunity to pass this data to the node constructor when constructed from an XML description, so I'm using the blackboard. For the second kind, I agree that the nodes should be reading the values on each tick(). The kind of data I'm using at this point is the former. So, an onInit-style method, called by the framework after construction and before the first tick, allows me to further initialize the node, if necessary, such as fetching data passed on the blackboard. However, I can work around this myself as I mentioned (by having a boolean variable and checking if it's the first iteration), but having the framework support it seems more elegant to me.

facontidavide, 2018-12-10:
> However, I can work around this myself as I mentioned (by having a boolean variable and checking if it's the first iteration), but having the framework support it seems more elegant to me.

I am all about elegance ;)
I will add onInit() tomorrow.

mjeronimo, 2018-12-11:
Great! By the way, we're doing this for both Conditions and Actions. It would be nice if it was supported for all of the node types.

facontidavide, 2018-12-11:
Can you please take a look to this branch before I merge it?

https://github.com/BehaviorTree/BehaviorTree.CPP/tree/issue_33

You will notice that I decided to throw an exception if someone tries to access the blackboard in the constructor.
I want people to be aware that it is a logical error and that there is another (correct) way to do it.

mjeronimo, 2018-12-11:
@facontidavide Ah, I didn't realize that! Thanks for the suggestion; I'll give it a try. That should be sufficient for me if you want to avoid introducing onInit.

mjeronimo, 2018-12-11:
@facontidavide I tried using the new BT code with our code. I'm getting an exception on onInit() when trying to use the blackboard (blackboard()->get<>). 

Besides the initializing parameters (which I can move to constructors as you've suggested), there is setup of shared shared variables going on in onInit that requires access to the blackboard. For example, setting up a path variable that will be used by the global planner (as output) and local planner (as input). Basically, blackboard() thinks it is in the constructor, but we're actually in onInit(). For example, my code looks something like this:

```
void onInit() override
{
  auto value = blackboard()->get<type>("key");
}
```

But the call to blackboard fails with the exception thrown ("You can NOT access the blackboard in the constructor").

facontidavide, 2018-12-12:
Ok, I will double check, I probably messed out

facontidavide, 2018-12-12:
Well, my bad for not writing a unit test first ;)

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/issue_33/gtest/gtest_blackboard.cpp

Now I am positive that it should work which this modifications 18f43ce53b3d84b43a7e4b40a4ae4c910449e1cd

closed 2018-12-12 by facontidavide

mjeronimo, 2018-12-12:
@facontidavide Thanks! I tried the latest and it worked; no issues.