BehaviorTree/BehaviorTree.CPP #34: relax requirements of convertFromString

v-lopez, 2018-12-11:
Don't fail compilation, send a warning instead

closed 2018-12-11 by facontidavide

facontidavide, 2018-12-11:
@mjeronimo for your information:

I decided to be less strict regarding convertFromString.

Now the program will compile even if the template specialization is missing.
If the NodeParameter points to a Blackboard, everything will work just fine since no string parsing is needed.

If either:

- NodeParameter was actually a string, not a pointer to a Blackboard entry __or__
- The Blackboard entry contains a string and you are trying to extract something else.

an exception is thrown, warning that you must implement your own __convertFromString__.

mjeronimo, 2018-12-11:
@facontidavide Sounds reasonable. Thanks for the info.