BehaviorTree/BehaviorTree.CPP #38: MinitraceLogger throws std::logic_error

Thordreck, 2018-12-13:
Hi,
I just noticed that the `MinitraceLogger` throws an exception complaining that only one instance is allowed, even if all the previous instances were deleted. I think the reference variable should be set to false in the destructor. From `bt_minitrace_logger.cpp` file (the exception message seems to be a bit off too):

```
MinitraceLogger::MinitraceLogger(TreeNode* root_node, const char* filename_json)
  : StatusChangeLogger(root_node)
{
    bool expected = false;
    if (!ref_count.compare_exchange_strong(expected, true))
    {
        throw std::logic_error("Only one instance of StdCoutLogger shall be created");
    }

    minitrace::mtr_register_sigint_handler();
    minitrace::mtr_init(filename_json);
    this->enableTransitionToIdle(true);
}

MinitraceLogger::~MinitraceLogger()
{
    minitrace::mtr_flush();
    minitrace::mtr_shutdown();
}
```

Cheers.

facontidavide, 2018-12-13:
you are absolutely right, thanks for reporting. I will fix this ASAP

facontidavide, 2018-12-13:
i believe it should work now

closed 2018-12-13 by facontidavide

bhagwanalbert, 2019-07-29:
Hi,

Something similar is happening to me with PublisherZMQ. I have the same code for all the loggers and this one does not work even if it is deleted. Only when the ROS node dies it lets me use this publisher in order to view the new tree in Groot.

Thanks in advance.