BehaviorTree/BehaviorTree.CPP #45: Subtree parameters

Masadow, 2019-01-07:
If there is, then I'm unaware (and it's not possible to add some in Groot) but that would be pretty neat to add the possibility to give parameters to subtree.

For instance, I have a subtree called Walk that is a sequence of nodes FaceTo and MoveTo.

Actually, I have to specify the target in both FaceTo and MoveTo while I would like to specify it at Walk level. Then FaceTo and MoveTo would just have to read Walk target value.

facontidavide, 2019-01-08:
I agree that adding parameters (ports in version 3) to Subtree avoid name collision and allow reusability, but I am puzzled by this sentence:

> In fact, I have to make a new subtree each time I want to face a different target.

That's not the way it works:

Suppose that you have 

```XML
<BehaviorTree ID="Walk">
<SequenceStar>
   <MoveTo target="{my_target}" />
   <FaceTo target="{my_target}" />
</SequenceStar>
</BehaviorTree>
```

You just have to visit multiple targets (let's say 10 targets), instead of creating "10 subtrees with different targets" I would do something like:

 ```XML
<BehaviorTree ID="VisitManyLocations">
    <RepeatUntilEmptyFIFO queue={targets_list}>
        <SequenceStar>
            <PullFromFIFO queue={targets_list} output_target="{my_target}" />
            <Walk/>
        </SequenceStar>
    </RepeatUntilEmptyFIFO>
</BehaviorTree>
```
Note: 
- RepeatUntilEmptyFIFO is a custom DecoratorNode.
- PullFromFIFO a custom SyncActionNode.
- Walk is the subtree we specified earlier.

Nevertheless, this will not solve completely the issue I am discussin in #44

facontidavide, 2019-01-17:
I am currently working on this very feature... stay tuned :)

facontidavide, 2019-01-25:
I created a tutorial to introduce this new feature.
Let me know if it is sufficiently self clear. https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ver_3/examples/t07_subtree_port_remapping.cpp