BehaviorTree/BehaviorTree.CPP #46: ROS package folder layout levels

bobhendrikx, 2019-01-14:
I am trying to use the ROS package installed using `apt-get install ros-kinetic-behaviortree-cpp`,
and I am having issues with the includes. In my CMakeLists i have:

```
find_package(catkin REQUIRED COMPONENTS behaviortree_cpp)
add_executable(bt_test test/bt_test.cpp )
target_link_libraries(bt_test ${BEHAVIOR_TREE_LIBRARY} )
include_directories(bt_test ${behaviortree_cpp_INCLUDE_DIRS})
```

To compile my files i need in my cpp:
```
#include "behaviortree_cpp/behaviortree_cpp/behavior_tree.h"
#include "behaviortree_cpp/behaviortree_cpp/bt_factory.h"
#include "behaviortree_cpp/behaviortree_cpp/basic_types.h"
```
Since also the files are installed in the path `/opt/ros/kinetic/include/behaviortree_cpp/behaviortree_cpp`
However, i get:
```
/opt/ros/kinetic/include/behaviortree_cpp/behaviortree_cpp/behavior_tree.h:17:53: fatal error: behaviortree_cpp/controls/parallel_node.h: No such file or directory
```
It appears that the includes in the library headers are missing a folder level of `behaviortree_cpp`.
prefixing with `behaviortree_cpp` in the library includes appears to solve the problem. 
Or am I doing something wrong in my CMakeLists here?

facontidavide, 2019-01-14:
Thank you for reporting. I will check it out and fix it ASAP

facontidavide, 2019-01-14:
Thanks for reporting. It is fixed now. I am releasing the new version 2.5.1 for all ROS1 distros.

closed 2019-01-14 by facontidavide