BehaviorTree/BehaviorTree.CPP #52: ver_3 coroutines segfault and other issues

v-lopez, 2019-01-30:
In this [code](https://gist.github.com/v-lopez/e4c3bf8e2075af66d9d22862225ef53d) there are 2 tests. 

1. `do_action` which will execute a CoRoutine several times, it is there to illustrate the need to manually perform `node.setStatus(BT::NodeStatus::IDLE);` if we want the action to be evaluated again. I'm not sure what is the proper way of executing a test similar to this without having to include this hard coded fix.

2. `do_action_timeout` Same action but with a timeout, when halt is called there's a segfault.

Here's the callstack for the segfault:
```
Stack trace (most recent call last) in thread 8354:
#15   Object "", at 0xffffffffffffffff, in 
#14   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 109, in __clone [0x7f8ea758d41c]
#13   Source "/build/glibc-Cl5G7W/glibc-2.23/nptl/pthread_create.c", line 333, in start_thread [0x7f8ea72706b9]
#12   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7f8ea7b1ec7f, in 
#11   Source "/usr/include/c++/5/thread", line 115, in _M_run [0x7f8ea81858d7]
        112: 	{ }
        113: 
        114: 	void
      > 115: 	_M_run() { _M_func(); }
        116:       };
        117: 
        118:   private:
#10   Source "/usr/include/c++/5/functional", line 1520, in operator() [0x7f8ea8185947]
       1517:       operator()()
       1518:       {
       1519:         typedef typename _Build_index_tuple<sizeof...(_Args)>::__type _Indices;
      >1520:         return _M_invoke(_Indices());
       1521:       }
       1522: 
       1523:     private:
#9    Source "/usr/include/c++/5/functional", line 1531, in _M_invoke<> [0x7f8ea81859f1]
       1528: 	  // std::bind always forwards bound arguments as lvalues,
       1529: 	  // but this type can call functions which only accept rvalues.
       1530:           return std::forward<_Callable>(std::get<0>(_M_bound))(
      >1531:               std::forward<_Args>(std::get<_Indices+1>(_M_bound))...);
       1532:         }
       1533: 
       1534:       std::tuple<_Callable, _Args...> _M_bound;
#8    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/include/behaviortree_cpp/decorators/timer_queue.h", line 70, in TimerQueue [0x7f8ea8180d04]
         67:   public:
         68:     TimerQueue()
         69:     {
      >  70:         m_th = std::thread([this] { run(); });
         71:     }
         72: 
         73:     ~TimerQueue()
#7    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/include/behaviortree_cpp/decorators/timer_queue.h", line 188, in run [0x7f8ea8181713]
        186:             // Check and execute as much work as possible, such as, all expired
        187:             // timers
      > 188:             checkWork();
        189:         }
        190: 
        191:         // If we are shutting down, we should not have any items left,
#6    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/include/behaviortree_cpp/decorators/timer_queue.h", line 228, in checkWork [0x7f8ea8181c39]
        226:             lk.unlock();
        227:             if (item.handler)
      > 228:                 item.handler(item.id == 0);
        229:             lk.lock();
        230:         }
        231:     }
#5    Source "/usr/include/c++/5/functional", line 2267, in operator() [0x7f8ea81825fd]
       2264:     {
       2265:       if (_M_empty())
       2266: 	__throw_bad_function_call();
      >2267:       return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
       2268:     }
       2269: 
       2270: #if __cpp_rtti
#4    Source "/usr/include/c++/5/functional", line 1871, in _M_invoke [0x7f8ea8180592]
       1868:       static void
       1869:       _M_invoke(const _Any_data& __functor, _ArgTypes&&... __args)
       1870:       {
      >1871: 	(*_Base::_M_get_pointer(__functor))(
       1872: 	    std::forward<_ArgTypes>(__args)...);
       1873:       }
       1874:     };
#3    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/decorators/timeout_node.cpp", line 55, in operator() [0x7f8ea8180159]
         52:             timer_id_ = timer().add(std::chrono::milliseconds(msec_), [this](bool aborted) {
         53:                 if (!aborted && child()->status() == NodeStatus::RUNNING)
         54:                 {
      >  55:                     child()->halt();
         56:                     child_halted_ = true;
         57:                 }
         58:             });
#2    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/behaviortree_ros_actions/test/simple_coro_test.cpp", line 22, in halt [0x422fa9]
         19:   virtual void halt() override
         20:   {
         21:     std::cout << "Action was halted" << std::endl;
      >  22:     BT::CoroActionNode::halt();
         23:   }
         24: 
         25: protected:
#1    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/action_node.cpp", line 202, in halt [0x7f8ea8137b55]
        199: {
        200:     if( _p->coro != 0 )
        201:     {
      > 202:         coroutine::destroy(_p->coro);
        203:         _p->coro = 0;
        204:     }
        205: }
#0    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/3rdparty/coroutine/coroutine.h", line 296, in destroy [0x7f8ea8138f4c]
        294: inline void destroy(routine_t id)
        295: {
      > 296: 	Routine *routine = ordinator.routines[id-1];
        297: 	assert(routine != nullptr);
        298: 
        299: 	delete routine;
Segmentation fault (Address not mapped to object [(nil)])
Segmentation fault
```

facontidavide, 2019-01-31:
Thank you, there were issues both in coroutine and timeout.

I improved and included your unit test. 
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ver_3/gtest/gtest_coroutines.cpp

There is one thing in particular that you should notice:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ver_3/gtest/gtest_coroutines.cpp#L27

You forgot to do the cleanup of start_time_.

By the way, I also tried to fix the issue with the fact that IDLE must be set manually in test like this one.

facontidavide, 2019-01-31:
I realized it is not solved yet.. More work to fix this

facontidavide, 2019-01-31:
ok, NOW it is solved.

For your information...
The problem was that I can not destroy a coroutine from a thread different from the one that created it.
This might be a problem with the 3rd party implementation of coroutine I am using, that is pretty basic.

Any way, the timeout execute the CoroActionNode::halt() from the thread of the timer, triggering the error.

Nice catch!! Thanks for the help.

commit 9efa8dbd (2019-01-31): removing wrong "solution". This should work to fix #52
closed 2019-01-31 by facontidavide