BehaviorTree/BehaviorTree.CPP #54: Updated release to ROS2 build farm?

mjeronimo, 2019-02-11:
@facontidavide 
We'd like to coordinate with you on release timing. Do you have plans for a follow-up release of the latest code?

facontidavide, 2019-02-11:
Sure. version 3.0 is in my opinion ready.

I am currently focusing on Groot that is out of date... a lot!

When are you doing your next release? Any target date?

mjeronimo, 2019-02-14:
We are targeting a code freeze for May 9th and a release for June 15th to coincide with the ROS2 D Turtle release. So, an update by mid-March would be good.

facontidavide, 2019-02-14:
Mid March will be perfect for me

mkhansen-intel, 2019-02-14:
Can we clarify, I'm slightly confused by this thread.

We want to do a Crystal update, I think that's the mid-March update @mjeronimo is referring to?
Then we'll do a "Dashing" release, which is code freeze early/mid May and release May 31st. 

I think the 3.0 version you're referring to is probably better aligned with the "Dashing" release, unless it is backward compatible?

I was thinking the mid-March update would be the 2.5.1 or similar (2.6.0?)release. We have some updates in PRs that are waiting on that release.

mjeronimo, 2019-02-14:
Davide's recent changes to the BehaviorTree.CPP library will require some adaptation on our part. The gist of this thread was to coordinate an update from him with our own development work. In particular, to give us  some time to make our changes before our Dashing release. I don't think it's necessary to include these changes for a Crystal update. The PRs that we have don't depend on these changes. Carlos has used a feature that will be included in the new library, but can easily avoid that the incorporate his changes without needing an update to the BT library (we've discussed this).

mkhansen-intel, 2019-02-14:
OK, thanks for clarifying Mike. I understand now.

facontidavide, 2019-03-01:
Hi,

eventually we have V3 ready to be used and well documented.
You can find the migration guide here: https://behaviortree.github.io/BehaviorTree.CPP/MigrationGuide/

Nevertheless, I will be happy to help migrating the code myself. If you need we can chat or have a quick telco the next week.

Cheers

hlzl, 2019-03-02:
> Hi,
>
> eventually we have V3 ready to be used and well documented.
> You can find the migration guide here: https://behaviortree.github.io/BehaviorTree.CPP/MigrationGuide/
>
> Nevertheless, I will be happy [...]

Thanks for the great work on ver_3!
I just read through the tutorial and migration guide and was wondering if the behavior of the Sequence Node described in tutorial 4 is now actually the behavior of the ReactiveSequence Node?

facontidavide, 2019-03-02:
@hlzl  Thanks for reporting, I will check that out

mjeronimo, 2019-03-06:
Thanks Davide. I will read the migration guide now and set about porting our code. I'll let you know if/when I have any questions.

facontidavide, 2019-03-13:
I just set a pull request to release version 3.0.4 of the library.

https://github.com/ros/rosdistro/pull/20558

I am not going to be available for few days, but we may talk the next week

mkhansen-intel, 2019-03-13:
@mjeronimo and @facontidavide - does 3.0.4 have API breaking changes that will require us to change to use it? I thought we were going to hold off on that until the Dashing release?

mjeronimo, 2019-03-13:
@mkhansen-intel Yes, 3.0.x has breaking changes. We agreed that Davide would update his library mid-March, to give us time to make the required changes before our release.

mkhansen-intel, 2019-03-15:
Since the 3.0 release was created as a new package name, we're fine. When we migrate to using it we'll need to update our dependencies to point to the new package. So everything looks good.

closed 2019-11-21 by facontidavide