BehaviorTree/BehaviorTree.CPP #57: Race condition in timeout decorator node?

Thordreck, 2019-02-22:
Hi,
Sorry if this explanation sounds a bit convoluted, but we have been running into this intermittent issue for a while now. A **sequence** followed by a **timeout** connected to a **retry** node would sometimes fail with an exception stating that "This shouldn't happen". The **timeout** node was somehow returning IDLE to its parent.
I took a look, and it seems the timer runs in its own thread, halting its child and setting a flag if the timeout ever kicks in. This flag is later checked in the tick function.
On the other hand, the **retry** node returns its own status by calling status() if its child returns FAILURE and there are still tries left.
I think what's happening is that the timer kicks in and halts the **retry** node at the same time its tick function is being called, thus making it return IDLE at the end.
What do you think? Is this possible?

facontidavide, 2019-02-22:
Thanks for reporting, I will try to fix this as soon as possible

facontidavide, 2019-02-25:
I would like to reproduce this as a unit test... any suggestion?

facontidavide, 2019-02-25:
ok, I think I fixed it

closed 2019-02-25 by facontidavide

Thordreck, 2019-02-25:
Sorry for the late answer. Thank you very much for the quick fix. I haven't had time to test it yet, but I'll check it out in the following days. Just for the record, we did this quick fix on our side (until we merge your changes):
```
NodeStatus TimeoutNode::tick()
{
    if( read_parameter_from_blackboard_ )
    {
        if( !getParam("msec", msec_) )
        {
            throw std::runtime_error("Missing parameter [msec] in TimeoutNode");
        }
    }

    if (status() == NodeStatus::IDLE)
    {
        setStatus(NodeStatus::RUNNING);
        timeout_reached_ = false;

        if (msec_ > 0)
        {
            timer().add(std::chrono::milliseconds(msec_), [this](bool aborted) {
                if (!aborted && child()->status() == NodeStatus::RUNNING)
                {
                    timeout_reached_ = true;
                }
            });
        }
    }

    const auto child_status = child()->executeTick();
    if(timeout_reached_)
    {
        child()->halt();
        setStatus(NodeStatus::FAILURE);
        return status();
    }

    if (child_status != NodeStatus::RUNNING)
    {
        timer().cancelAll();
    }

    setStatus(child_status);
    return status();
}

void TimeoutNode::halt()
{
    timer().cancelAll();
    DecoratorNode::halt();
}
```