BehaviorTree/BehaviorTree.CPP #59: Desired Behavior RetryUntilSuccessful Node

hlzl, 2019-02-26:
When describing the RetryUntilSuccesful Node on the BehaviorTree.CPP [page](https://behaviortree.github.io/BehaviorTree.CPP/BT_basics/) it is stated that the Retry node will repeat ticking the __child__ up to N times if the child returns FAILURE. Currently however, if the child returns FAILURE the whole tree starts again from root. This could cause problems or at least be inefficient e.g. if a robot first moves back to an old position before going to the current position and trying an action again. So, I was wondering if the latter is the desired behavior of RetryUntilSuccessful?

facontidavide, 2019-02-26:
First of all, I care more about the semantic than the efficiency, but I see your point.

According to version 3 (the only I care about right now):

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ver_3/src/decorators/retry_node.cpp

If a child returns FAILURE, then `RetryNode` returns either:
- RUNNING if attempts < N
- FAILURE if attempts == N

The real question is: 

Suppose we have a `RetryNode(N)` with a __synchronous__ action as child.
Do we want to be able to interrupt the Retry-loop between an execution of the Action and the next one?

Different people might answer in different way, unfortunately.

facontidavide, 2019-02-27:
For consistency, I decided to apply the changes you mentioned.

Now:

- Retry is equivalent to a Fallback with N instances of the same child
- Repeat is equivalent to a Sequence with N instances of the same child

closed 2019-02-27 by facontidavide