BehaviorTree/BehaviorTree.CPP #63: Building BehaviorTree.CPP (Visual Studio 2017/Windows 10 x64)

DJuego, 2019-03-02:
Hi! Thank you for your fantastic contribution, and congrats for your new shiny version 3.0!

I am a great fan of crossplatform, minimal dependencies and header-only gems in modern c++.

Well; BehaviourTree.cpp is not header-only :-). So i am trying to compile/build the library from source code, in Visual Studio 2017(15.9.7) for Windows 10 x64. No success. :-/. At the moment i have omitted GTEST amd ZMQ because i only want the core/logic functionality. Is it possible? 

Any guide/hits about how to compile/build/configure BehaviorTree.CPP for using in new proyects? Thanks in advance!! 

I have enabled the _CRT_SECURE_NO_WARNINGS flag.

I get this output :-(:

```

1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: Debug x64 ------
1>Checking Build System
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/debug/CMakeFiles/generate.stamp is up-to-date.
2>------ Rebuild All started: Project: behaviortree_cpp_v3, Configuration: Debug x64 ------
2>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/CMakeLists.txt
2>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/debug/CMakeFiles/generate.stamp is up-to-date.
2>action_node.cpp
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(115): error C2220: warning treated as error - no 'object' file generated
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(115): warning C4267: 'return': conversion from 'size_t' to 'coroutine::routine_t', possible loss of data
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(137): warning C4100: 'lpParameter': unreferenced formal parameter
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(788): error C2059: syntax error: '('
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(788): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(788): error C2059: syntax error: ')'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(789): warning C4183: '__attribute__': missing return type; assumed to be a member function returning 'int'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(795): error C3861: 'backtrace': identifier not found
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2589: '(': illegal token on right side of '::'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2144: syntax error: 'unknown-type' should be preceded by ')'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2661: 'std::vector<void *,std::allocator<_Ty>>::resize': no overloaded function takes 0 arguments
2>        with
2>        [
2>            _Ty=void *
2>        ]
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2144: syntax error: 'unknown-type' should be preceded by ';'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2143: syntax error: missing ')' before '.'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C3928: '->': trailing return type is not allowed after a parenthesized declarator
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C3484: syntax error: expected '->' before the return type
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C3613: missing return type after '->' ('int' assumed)
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): warning C4458: declaration of '_stacktrace' hides class member
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(679): note: see declaration of 'backward::StackTraceImplHolder::_stacktrace'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2146: syntax error: missing ';' before identifier 'size'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\backward-cpp/backward.hpp(812): error C2059: syntax error: ')'
2>basic_types.cpp
2>behavior_tree.cpp
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/tree_node.h(210): error C2220: warning treated as error - no 'object' file generated
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/tree_node.h(210): warning C4127: conditional expression is constant
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/tree_node.h(211): note: consider using 'if constexpr' statement instead
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/actions/set_blackboard_node.h(49): note: see reference to function template instantiation 'BT::Result BT::TreeNode::getInput<std::string>(const std::string &,T &) const' being compiled
2>        with
2>        [
2>            T=std::string
2>        ]
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/blackboard.h(170): error C2664: 'std::string BT::demangle(const char *)': cannot convert argument 1 from 'const type_info *const ' to 'const char *'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/blackboard.h(170): note: Types pointed to are unrelated; conversion requires reinterpret_cast, C-style cast or function-style cast
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/tree_node.h(258): note: see reference to function template instantiation 'void BT::Blackboard::set<T>(const std::string &,const T &)' being compiled
2>        with
2>        [
2>            T=std::string
2>        ]
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/actions/set_blackboard_node.h(57): note: see reference to function template instantiation 'BT::Result BT::TreeNode::setOutput<std::string>(const std::string &,const T &)' being compiled
2>        with
2>        [
2>            T=std::string
2>        ]
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/blackboard.h(171): error C2664: 'std::string BT::demangle(const char *)': cannot convert argument 1 from 'const type_info' to 'const char *'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\include\behaviortree_cpp/blackboard.h(171): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
2>blackboard.cpp
[... 300 lines cut ...]
2>tinyxml2.cpp
2>minitrace.cpp
2>p:\compilados\x32-x64\trabajo_msvc2017_x64\behaviortree.cpp\3rdparty\minitrace\minitrace.h(232): error C2220: warning treated as error - no 'object' file generated
2>p:\compilados\x32-x64\trabajo_msvc2017_x64\behaviortree.cpp\3rdparty\minitrace\minitrace.h(232): warning C4244: '=': conversion from 'int64_t' to 'double', possible loss of data
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\minitrace\minitrace.cpp(87): warning C4244: 'initializing': conversion from 'double' to 'int64_t', possible loss of data
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\minitrace\minitrace.cpp(280): warning C4456: declaration of 'len' hides previous local declaration
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\minitrace\minitrace.cpp(241): note: see declaration of 'len'
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\minitrace\minitrace.cpp(281): warning C4456: declaration of 'i' hides previous local declaration
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\minitrace\minitrace.cpp(228): note: see declaration of 'i'
[... cut ...]
3>------ Rebuild All started: Project: ALL_BUILD, Configuration: Debug x64 ------
3>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/CMakeLists.txt
3>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/debug/CMakeFiles/generate.stamp is up-to-date.
4>------ Skipped Rebuild All: Project: INSTALL, Configuration: Debug x64 ------
4>Project not selected to build for this solution configuration 
========== Rebuild All: 2 succeeded, 1 failed, 1 skipped ==========
```

facontidavide, 2019-03-02:
Hi @DJuego 

I am a fan of cross platform too. Let's try to fix this (I will also add an Appveyor CI to be sure that it always compile in Windows).

Neither ZMQ nor GTEST should be mandatory.

I will try to fix this the next week.

lukeliao, 2019-03-06:
Hi, I am also interested in this cross platform part.
I plan to use this library to deal with some complicated robot manipulation task.
Very glad to see your progress and thank you for your contribution!!

facontidavide, 2019-03-06:
Working on it...

facontidavide, 2019-03-08:
well, THAT was a lot of work. Let me know if everything works

DJuego, 2019-03-08:
Thank you very much  for your great efforts! I am very grateful.

Unfortunately, there are problems with the Release configuration... I got:
```
1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: Release Win32 ------
1>Checking Build System
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/tools/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/examples/CMakeFiles/generate.stamp is up-to-date.
2>------ Rebuild All started: Project: behaviortree_cpp_v3, Configuration: Release Win32 ------
2>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/CMakeLists.txt
2>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/CMakeFiles/generate.stamp is up-to-date.
2>action_node.cpp
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(180): error C2220: warning treated as error - no 'object' file generated
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(180): warning C4189: 'routine': local variable is initialized but not referenced
2>basic_types.cpp
2>behavior_tree.cpp
2>blackboard.cpp
2>bt_factory.cpp
2>decorator_node.cpp
2>condition_node.cpp
2>control_node.cpp
2>shared_library.cpp
2>tree_node.cpp
2>xml_parsing.cpp
2>inverter_node.cpp
2>repeat_node.cpp
2>retry_node.cpp
2>subtree_node.cpp
2>timeout_node.cpp
2>fallback_node.cpp
2>parallel_node.cpp
2>reactive_sequence.cpp
2>reactive_fallback.cpp
2>Generating Code...
2>Compiling...
2>sequence_node.cpp
2>sequence_star_node.cpp
2>bt_cout_logger.cpp
2>bt_file_logger.cpp
2>bt_minitrace_logger.cpp
2>tinyxml2.cpp
2>minitrace.cpp
2>backward.cpp
2>shared_library_WIN.cpp
2>Generating Code...
2>Done building project "behaviortree_cpp_v3.vcxproj" -- FAILED.
3>------ Rebuild All started: Project: crossdoor_nodes, Configuration: Release Win32 ------
4>------ Rebuild All started: Project: dummy_nodes, Configuration: Release Win32 ------
5>------ Rebuild All started: Project: movebase_node, Configuration: Release Win32 ------
6>------ Rebuild All started: Project: bt_log_cat, Configuration: Release Win32 ------
7>------ Rebuild All started: Project: bt_plugin_manifest, Configuration: Release Win32 ------
8>------ Rebuild All started: Project: crossdoor_nodes_dyn, Configuration: Release Win32 ------
9>------ Rebuild All started: Project: dummy_nodes_dyn, Configuration: Release Win32 ------
10>------ Rebuild All started: Project: movebase_node_dyn, Configuration: Release Win32 ------
3>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
5>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
7>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/tools/CMakeLists.txt
9>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
6>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/tools/CMakeLists.txt
4>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
8>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
5>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
3>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
6>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/tools/CMakeFiles/generate.stamp is up-to-date.
9>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
7>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/tools/CMakeFiles/generate.stamp is up-to-date.
8>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
4>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
[... 126 lines cut ...]
15>------ Build started: Project: t04_reactive_sequence, Configuration: Debug Win32 ------
16>------ Build started: Project: t05_crossdoor, Configuration: Debug Win32 ------
17>------ Build started: Project: t06_subtree_port_remapping, Configuration: Debug Win32 ------
18>------ Build started: Project: t07_wrap_legacy, Configuration: Debug Win32 ------
11>t01_first_tree_dynamic.vcxproj -> P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\builds\examples\Debug\t01_first_tree_dynamic.exe
[... cut ...]
12>t01_first_tree_static.vcxproj -> P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\builds\examples\Debug\t01_first_tree_static.exe
21>t10_include_trees.vcxproj -> P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\builds\examples\Debug\t10_include_trees.exe
22>------ Build started: Project: ALL_BUILD, Configuration: Debug Win32 ------
23>------ Skipped Build: Project: INSTALL, Configuration: Debug Win32 ------
23>Project not selected to build for this solution configuration 
========== Build: 22 succeeded, 0 failed, 0 up-to-date, 1 skipped ==========
```

DJuego

facontidavide, 2019-03-08:
try now

DJuego, 2019-03-08:
Wops! It works! 

Debug Win32
Release Win32

Congrats! 

However, actually, i work with the x64 :-/ Sorry! 

So I create a Debug x64 and Release x64 configurations. I copy settings from Win32.

When i build the Debug x64 configuration i get:

```
1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: Debug x64 ------
1>Checking Build System
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/tools/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/sample_nodes/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/examples/CMakeFiles/generate.stamp is up-to-date.
2>------ Rebuild All started: Project: behaviortree_cpp_v3, Configuration: Debug x64 ------
2>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/CMakeLists.txt
2>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/CMakeFiles/generate.stamp is up-to-date.
2>action_node.cpp
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(115): error C2220: warning treated as error - no 'object' file generated
2>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\3rdparty\coroutine/coroutine.h(115): warning C4267: 'return': conversion from 'size_t' to 'coroutine::routine_t', possible loss of data
2>basic_types.cpp
2>behavior_tree.cpp
2>blackboard.cpp
2>bt_factory.cpp
2>decorator_node.cpp
2>condition_node.cpp
2>control_node.cpp
2>shared_library.cpp
2>tree_node.cpp
2>xml_parsing.cpp
2>inverter_node.cpp
2>repeat_node.cpp
2>retry_node.cpp
2>subtree_node.cpp
2>timeout_node.cpp
2>fallback_node.cpp
2>parallel_node.cpp
2>reactive_sequence.cpp
2>reactive_fallback.cpp
2>Generating Code...
2>Compiling...
2>sequence_node.cpp
2>sequence_star_node.cpp
2>bt_cout_logger.cpp
2>bt_file_logger.cpp
2>bt_minitrace_logger.cpp
2>tinyxml2.cpp
2>minitrace.cpp
2>backward.cpp
2>shared_library_WIN.cpp
2>Generating Code...
2>Done building project "behaviortree_cpp_v3.vcxproj" -- FAILED.
3>------ Rebuild All started: Project: crossdoor_nodes, Configuration: Debug x64 ------
4>------ Rebuild All started: Project: dummy_nodes, Configuration: Debug x64 ------
5>------ Rebuild All started: Project: movebase_node, Configuration: Debug x64 ------
6>------ Rebuild All started: Project: bt_log_cat, Configuration: Debug x64 ------
7>------ Rebuild All started: Project: bt_plugin_manifest, Configuration: Debug x64 ------
8>------ Rebuild All started: Project: crossdoor_nodes_dyn, Configuration: Debug x64 ------
9>------ Rebuild All started: Project: dummy_nodes_dyn, Configuration: Debug x64 ------
10>------ Rebuild All started: Project: movebase_node_dyn, Configuration: Debug x64 ------
4>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
3>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/sample_nodes/CMakeLists.txt
9>C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(391,5): warning MSB8028: The intermediate directory (x64\Debug\) contains files shared from another project (crossdoor_nodes.vcxproj, dummy_nodes.vcxproj).  This can lead to incorrect clean and rebuild behavior.
10>C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(391,5): warning MSB8028: The intermediate directory (x64\Debug\) contains files shared from another project (crossdoor_nodes.vcxproj, dummy_nodes.vcxproj).  This can lead to incorrect clean and rebuild behavior.
[... 113 lines cut ...]
19>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\examples\t04_reactive_sequence.cpp : fatal error C1041: cannot open program database 'P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\builds\examples\x64\Debug\vc141.pdb'; if multiple CL.EXE write to the same .PDB file, please use /FS
19>Done building project "t04_reactive_sequence.vcxproj" -- FAILED.
21>t10_include_trees.cpp
21>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\examples\t10_include_trees.cpp : fatal error C1041: cannot open program database 'P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\builds\examples\x64\Debug\vc141.pdb'; if multiple CL.EXE write to the same .PDB file, please use /FS
21>Done building project "t10_include_trees.vcxproj" -- FAILED.
16>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/examples/CMakeFiles/generate.stamp is up-to-date.
18>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/examples/CMakeFiles/generate.stamp is up-to-date.
16>t01_build_your_first_tree.cpp
16>P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\examples\t01_build_your_first_tree.cpp : fatal error C1041: cannot open program database 'P:\Compilados\x32-x64\TRABAJO_MSVC2017_x64\BehaviorTree.CPP\builds\examples\x64\Debug\vc141.pdb'; if multiple CL.EXE write to the same .PDB file, please use /FS
18>t03_generic_ports.cpp
[... cut ...]
22>------ Rebuild All started: Project: ALL_BUILD, Configuration: Debug x64 ------
22>Building Custom Rule P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/CMakeLists.txt
22>CMake does not need to re-run because P:/Compilados/x32-x64/TRABAJO_MSVC2017_x64/BehaviorTree.CPP/builds/CMakeFiles/generate.stamp is up-to-date.
23>------ Skipped Rebuild All: Project: INSTALL, Configuration: Debug x64 ------
23>Project not selected to build for this solution configuration 
========== Rebuild All: 2 succeeded, 20 failed, 1 skipped ==========
´´´

Sorry again! 
DJuego

facontidavide, 2019-03-08:
For the time being I can't spend more time on this. I reopened the issue to deal with it in the future

lukeliao, 2019-03-09:
@DJuego Hi, I compile all code successfully. 
Hint:
Comment CMakeLists.txt line 165-169
Your problem is that compiler catch a warning in corountine.h and regard it as an error
And I have met with some problem using zmq in windows with vcpkg.
When I finish all problems, I will pull a request

DJuego, 2019-03-09:
Thank you, @lukeliao. That seems like good news. I have to admit that I'm too used to using header-only libraries with zero-dependencies. My fault. Portability and simplicity in integration are crucial for me. I only use CMake to compile tests or examples when necessary. 

DJuego

Masadow, 2019-03-10:
@facontidavide Would be nice if you could add the .sln or .vcxproj file somewhere so we don't have to configure everything by ourself :)

facontidavide, 2019-03-12:
@Masadow compilation will always be done with cmake, I am afraid .sln or .vcxproj  might go "out of synch" and give me extra headache...

@DJuego thanks for being patient. the problem has been finally fixed.

@lukeliao I haven't had time to check ZMQ on windows, yet. My primary platform is Linux.

closed 2019-03-12 by facontidavide

DJuego, 2019-03-12:
> @DJuego thanks for being patient. the problem has been finally fixed.

Indeed!. I can confirm it. Building is working. :-) Thanks!

However, I have problems and discrepancies (debug vs release) when executing any examples. I attached the console outputs in debug and release settings for all official examples.

For example; "t01_first_tree_static.exe" in debug produce:

[image]

t04_reactive_sequence.exe produce differents outputs in Debug and Release

etc

DEBUG:
[image]

RELEASE:
[image]

Maybe it has to do with gtest is disabled? Has anyone already experimented with the examples in Windows x64?

In any case, don't worry for now! Surely, when we proceed to debug the examples we will have more details.

Thank you very very very much, @facontidavide !! Your effort and concern to bring this library to users of your non-primary OS, compiler, etc. deserves our recognition.

DJuego