BehaviorTree/BehaviorTree.CPP #68: TinyXML symbols exported on VC++

Masadow, 2019-03-11:
When compiling under msvc, tiny xml symbols are exported which is conflicting for me because I'm already using tinyxml on my own.

facontidavide, 2019-03-11:
tell me if the last commit helps... even if I doubt it :(

I don't have much time to address this right bow, but I will eventually.

Masadow, 2019-03-11:
It's not compiling at all now, complaining about BEHAVIOR_TREE_LIBRARY requiring BEHAVIOR_TREE_LIBRARY_EXTRA that is not exported. I don't know much about cmake nor windows compilation but since it's a must for my project at this point, I'll try to look at it and hopefully fix it as quickly as possible.

facontidavide, 2019-03-12:
ok, that was not a good idea, I reverted the change....

facontidavide, 2019-03-12:
I believe that I might have solve the issue with commit 108cd60c8c9716a639bb686d1d4e498fb9dc3225

Check it out

Masadow, 2019-03-12:
It's fixed, good job !

closed 2019-03-12 by facontidavide