BehaviorTree/BehaviorTree.CPP #69: Error compiling in release mode under VC++

Masadow, 2019-03-11:
Hi,

When I try to compile my project under debug against debug library of BT.CPP, it works. If I try to compile under debug against release BT.CPP, it works too. But if I try to compile under release against release BT.CPP. I've got the following error during linking : 

Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in main.obj

facontidavide, 2019-03-11:
I have NO idea what is all about, sorry. I don't actually program in windows. I guess it is a problem of the CMAKE configuration.

facontidavide, 2019-03-12:
Just pushed a change that add the suffix "d" to a library built in debug mode, of it is common on Windows.
In this way you should be able to fix your issue more easily.

facontidavide, 2019-03-12:
It should be fixed now. Please check it out

Masadow, 2019-03-12:
I confirm it's good

Masadow, 2019-03-12:
@facontidavide Now it's the opposite, if I compile my project under debug mode using release build, it complain with the same exact message except that 0 and 2 values are interchanged

facontidavide, 2019-03-12:
your debug build should depend on the library called __behaviortree_cpp_v3d__ (suffix d) whilst the release should link with __behaviortree_cpp_v3__ (no suffi).

Can you send me a dummy example to reproduce the error?

Masadow, 2019-03-12:
I'll try to make a dummy project to reproduce the error. And I'm sorry but I disagree, I should never load the library in debug mode unless I actually need to debug the library.

facontidavide, 2019-03-12:
ok, you probably know best, if you are used to develop in VC++ ( I don't)

Masadow, 2019-03-12:
I don't know much about VC++ but it seems to defy the logic here, after some research on my own, it seems like it's bad to mix release/debug DLL on windows, so my bad on this one. It doesn't make sense to me that it works this way. What's weird is that I encounter this error in my main project but if I try to link against a dummy hello world, I don't have this issue at all and the only library producing the error is BT.CPP. I'll just stick with debug for debug and release for release then

closed 2019-03-12 by facontidavide