BehaviorTree/BehaviorTree.CPP #71: Should nodes loop internally or not?

maxpfingsthorn, 2019-03-12:
I noticed that all nodes with multiple children (sequence, fallback, reactive*, retry, repeat, etc.) loop over children internally with while's and for's in their implementation of tick(). I wonder if this is a design decision, and if yes, why is this the case? Due to perceived efficiency?

I would argue that for each executeTick() on the root node, only one leaf should have their tick() method called. This would imply descending into the tree from the root every executeTick(), but I believe this cost would be negligible. However, tick()-ing only one leaf per executeTick() iteration would be more in line with the conceptual behavior tree model and it would help scheduling/preempting a lot.

What do you think?

facontidavide, 2019-03-12:
Hi @maxpfingsthorn 

this is a very good question, and I am struggling to find the right answer myself, because anyone of my users has a different opinion.

I agree with you that semantic correctness should be preferable over perceived efficiency.

I believe too that avoiding the internal loop might be preferable because it gives more opportunities to the logic of the tree to preempt a branch.

Unfortunately, this is a quite important change that might break someone else code in some subtle way...

Summarizing, I do see your point and I actually share it, but such a change need a debate among users of these library; in such a debate, efficiency MUST not be considered, in my opinion, only correctness and predictability.

maxpfingsthorn, 2019-03-12:
Hi @facontidavide,

Ok, I see your point on this being a potentially breaking change. I would be happy to discuss.

What is a good forum to do this?

facontidavide, 2019-03-13:
No forum, unfortunately.
Take a loop to this branch: https://github.com/BehaviorTree/BehaviorTree.CPP/tree/preemptable_loop

Is this what you had in mind?

I realized that the Reactive ControlNodes cannot be modified. Take a look at the commit for more details: 2dc84b1886bdade079c9beab931ef4f93256b7a6

maxpfingsthorn, 2019-03-14:
That looks very good to me! Thank you.

For the reactive controls: Why not? The implementation would be the same as for the ones with memory, but would reset the index on RUNNING as well. Or did I miss something?

facontidavide, 2019-03-20:
@v-lopez @mjeronimo @miccol @Masadow @Thordreck

Would you like to join this discussion?

miccol, 2019-03-20:
I also struggled in the past to find the right answer. 
I guess here we are talking about asynchronous action nodes. I think in some practical cases we need to have actions that loop internally (e.g. a navigation action what waits in a loop until the goal is reached). 
The workaround was to check, at each loop, if the action is halted or not.

facontidavide, 2019-03-20:
> I guess here we are talking about asynchronous action nodes. 

Quite the opposite, I am worry about **synchronous** nodes.

Consider for instance this tree

```XML 
<ReactiveSequence>
    <CheckBattery>
    <Repeat num_cycles="10">
          <DoSomething>
    </Repeat>
</ReactiveSequence>
```

Let's suppose that `DoSomething` is synchronous and requires 1 second to be completed. The condition `CheckBattery` becomes false after 5 seconds.

The entire sequence will __always__ take 10 seconds; this goes against the "Minimum WTF principle". 
[image]

Unfortunately, since `Repeat` loops internally, `CheckBattery`is invoked only once, not 10 times.

Allow me to use the word __"atomic"__ as synonym of "synchronous" to make my point.

If `Sequence`, `Fallback`, `Repeat` or `RetryUntilSuccesfull` contains __only__ atomic children, that ControlNode / DecoratorNode itself becomes atomic, i.e impossible to halt.

This is probably not what we want: in the previous example, a user would expect `CheckBattery` to be called many times.

The problem with the new semantic proposed by @maxpfingsthorn is that all those Conditions inside Reactive ControlNodes might be called MUCH more often, but this is the point of reactive trees, isn't it?

v-lopez, 2019-03-20:
I agree that the example you provided should call check battery several times.

In my mind, the only "blocking" nodes should be the the sync action nodes. Any intermediate decorator that needs to loop should return RUNNING.

hlzl, 2019-03-20:
What confused me the most in the beginning was that we constantly call `executeTick()` from the root node when we use asynchronous nodes. I thought that because of these internal loops we could call `executeTick()` only once in the beginning and let it propagate through all nodes of the tree with Sequences ticking asynchronous children internally while they run. 
As this is not the case, I would agree with @maxpfingsthorn and prefer to tick only one leaf per `executeTick()`. However, as @miccol said, this would require to make an exception for ReactiveSequences with asynchronous nodes.

Regarding your example, I don't see why you should use a ReactiveSequence if you're not using an asynchronous node. I would expect an implementation like this to achieve your desired behavior
```xml
<Repeat num_cycles="10">
    <Sequence>
        <CheckBattery>
        <DoSomething>
    </Sequence>
</Repeat>
```

One thing I think should also be mentioned here are the semantics of two asynchronous nodes inside a ReactiveSequence. This will currently result in an infinite loop. However, I think it would be an interesting tool to have, let's call it a _ParallelAsynchronousNode_, which allows to execute two or more ansynchronous nodes (only once) at the same time. However, I couldn't find any theory on this kind of behavior and its semantics would have to be discussed.

facontidavide, 2019-03-20:
hi @hlzl , thanks for joining the discussion.

About the need to executeTick() repeatedly, that is "by design", otherwise a tree is not preemptable. But I need to make it less confusing in the documentation.

The alternative tree you suggested would work. what worries me is how intuitive a solution is compared to another for a newbie. 

The problem with reactive nodes as you pointed out, is indeed that a ReactiveSequence or ReactiveFallback must not have more than an asynchronous child.
This is also a potential source of bugs for many people. 

In the past, I used NodeStatus::IDLE to avoid ticking twice already ticked Actions, but @v-lopez complained about that too.

I am now thinking which is the least of two evils....

mjeronimo, 2019-03-20:
As a relative newcomer to BehaviorTrees, the semantic difference between

```
<ReactiveSequence>
    <CheckBattery>
    <Repeat num_cycles="10">
          <DoSomething>
    </Repeat>
</ReactiveSequence>
```

and

```
<Repeat num_cycles="10">
    <Sequence>
        <CheckBattery>
        <DoSomething>
    </Sequence>
</Repeat>
```
is clear to me.

However, I find "ReactiveSequence or ReactiveFallback must not have more than one asynchronous child" confusing. I think of an action node simply returning its current state (RUNNING, etc.) and don't see why a parent node needs to know the internals details of an action node, such as whether it asynchronously communicating with another thread or process.

miccol, 2019-03-21:
> In my mind, the only "blocking" nodes should be the the sync action nodes. Any intermediate decorator that needs to loop should return RUNNING.

I agree. 

> The problem with reactive nodes as you pointed out, is indeed that a ReactiveSequence or ReactiveFallback must not have more than an asynchronous child.
> This is also a potential source of bugs for many people

I have a solution to this in mind. I will discuss it in another thread.

> In the past, I used NodeStatus::IDLE to avoid ticking twice already ticked Actions, but @v-lopez complained about that too.

Is the discussion about somewhere? I used to like the IDLE status

facontidavide, 2022-11-22:
changed in version 4.
I decided to allow some Control/Decorators to return RUNNING between one child and the next

closed 2022-11-22 by facontidavide