BehaviorTree/BehaviorTree.CPP #72: Port mismatch betwen sibling subtrees

v-lopez, 2019-03-13:
It looks like the tree construction and internal to external mapping is not working 100% well.

See the attached tree:
[sibling_subtrees.txt](https://github.com/BehaviorTree/BehaviorTree.CPP/files/2962180/sibling_subtrees.txt)

[image]

It's fairly simple, it executes twice the subtree "Subtree1", which checks that a param is 1.

The problem is that the second Subtree execution fails, because it can't read the input port of param.

I started digging and got up to the point where the blackboards are created, and it looks like the parent of the blackboard of the second subtree, is the first subtree's blackboard, not the root node's blackboard.

v-lopez, 2019-03-13:
Also here's a test for verifying it:
```
static const char* xml_text_sibling_subtrees = R"(
PUT THE TREE HERE
  )";

TEST(BehaviorTreeFactory, SiblingSubtree)
{
    BehaviorTreeFactory factory;
    CrossDoor::RegisterNodes(factory);

    Tree tree = factory.createTreeFromText(xml_text_sibling_subtrees);

    StdCoutLogger logger(tree);
    printTreeRecursively(tree.root_node);
    EXPECT_EQ(NodeStatus::SUCCESS, tree.root_node->executeTick());
    logger.flush();
}

```

facontidavide, 2019-03-13:
I will address this the next week

maxpfingsthorn, 2019-03-14:
I ran across this bug as well! Thanks for tracking it down.

facontidavide, 2019-03-14:
Sorry about the trouble. I am away from the office until Monday, afterward I will try to fix this

facontidavide, 2019-03-19:
fixed (and unit test added)

closed 2019-03-19 by facontidavide