BehaviorTree/BehaviorTree.CPP #76: Void as port type

Thordreck, 2019-03-26:
Hi, I guess this is more of a question than an actual issue, but I was not sure where else I could ask.
I'm working in the migration to v3, and I was running into some issues porting some functionality to the new typed port system. Some of the blocks in my library do not know on compilation time the type of its result/input. I'm aware of the benefits of enforcing a strict type check, but I think some applications may benefit of a more relaxed port policy sometimes. Taking a look at the code it seems setting the type to `void` when declaring the port disables type checking, if I'm not mistaken.
Can users rely on this feature? If so, wouldn't make more sense to set a type-erasure container as the "wildcard type" (for example BT::Any)?
Sorry if this was already discussed/documented. I couldn't find anything about it in the repo/docs.

facontidavide, 2019-07-30:
"Some of the blocks in my library do not know on compilation time the type of its result/input."

Since this is C++, your code should handle correctly the type.
Having a "wildcard type" is precisely against the philosophy of this library, for that reason.

What if you connect a Pose3D with a Path2D?

Said that, it might be possible (even if bad practice) to use a **type erased** InputPort connected to a strongly typed OutputPort.

For instance, we may connect a port of type Pose3D (output) to BT::Any (input) and the latter node is the one responsible to cast BT::Any back to the original type.

Unfortunately this makes impossible to use **convertFromString**, for obvious reasons.

What do you think?

Thordreck, 2019-08-01:
Hi,
Thank you for your reply. It's been a while since I posted this, and my understanding of my own issue has improved a bit since then.
Let me explain briefly my use case and how I ran into these type issues. I wrote an interface between ROS and BehaviorTree.CPP. Users can register publishers/subscribers for their own messages using some templatized classes offered by the library. In the case of subscribers, for example, received messages can be written into the blackboard either as they come (meaning the output port type is well known) or they can choose to serialize them into a more general structure. If they choose the latter, they have access to several more general-purpose blocks (that are not tied to a specific message type). This allows to build logic without having to write extra code/blocks for each custom message that may pop up. Any field in the structure can be accessed later with another block, using the field names of the original ROS message (for example, the x coordinate in a `geometry_msgs/PointStamped` could be accessed with `/point/x`).
Now, the problem I had with this approach is that, at the time of defining the ports, I don't know their type (aka, I don't know which field the user will fetch). Once the tree is running and the path has been provided I do have that info and the process is type safe.
So basically I would define the output port type as void, fetch some message field with, let's say type `float`, and try to connect it simultaneously to several blocks that may expect types like `double` or `string`. In v3 this will result in an error, stating that the type of a port shall not change. I ran into similar issues with more complex types, where I was aware of a possible conversion between the current blackboard entry type and a block's input, but the port system was not flexible enough to do anything about it.

>What if you connect a Pose3D with a Path2D?

While that example may seem obviously incorrect, there are other situations and types pairs that fit more into the "gray-zone" category, imo. I have come to understand that my issue was not with the strong-typed port system itself, but with the inability to define custom types conversions.
I have been working on this these last days and I have implemented a type conversion facility. Custom conversions from type T to type U can be registered by users from their plugins. Something like: `factory.registerTypeConverter<T,U>(Functor)`
When parsing the xml or running the tree, all the output types for a given blackboard entry are compared with the input types. Only if there is a pair whose types are not the same and no custom conversion has been provided an error will be raised.
You can give it a look in our fork if you want:
https://github.com/robotics-upo/BehaviorTree.CPP/blob/types_conversion/include/behaviortree_cpp/utils/types_converter.hpp

I would like to hear your opinion on this, if possible. It's a more general approach than the `convertFromString` template specialization (or the built-in numeric and string conversions offered by BT::Any). Also these types conversions could be queried in Groot too and messages/warnings could be displayed accordingly.

facontidavide, 2023-07-21:
closed, being old

closed 2023-07-21 by facontidavide