BehaviorTree/BehaviorTree.CPP #77: ROS specific tutorial/example

doisyg, 2019-03-28:
Hello,
Not sure it is the right place to ask, but is there any example of BehaviorTree.CPP interacting with ROS (1) somewhere? And even better a sample calling a ROS action?
I am looking for good practice sample to start using it.

facontidavide, 2019-03-28:
my fault, I should have done this earlier. My plan is to add such examples here.... https://github.com/BehaviorTree/BehaviorTree.ROS

doisyg, 2019-03-28:
Looking forward to it!
(in the meantime, any public repo using BehaviorTree.CPP with ROS, even not documented, will help)

bhagwanalbert, 2019-05-04:
Hello,
Your library has been recommended to us and our goal it to implement it with ROS nodes, so I am also interested in a basic example of how to combine these two. May I know approximately when we will be able to have one basic tutorial?
Thanks in advance,
Albert.

facontidavide, 2019-07-28:
This is the basic example you are looking for:

https://github.com/BehaviorTree/BehaviorTree.ROS/tree/master/src/actions

Cheers

closed 2019-07-28 by facontidavide

varunvp, 2019-08-03:
> This is the basic example you are looking for:
> 
> https://github.com/BehaviorTree/BehaviorTree.ROS/tree/master/src/actions
> 
> Cheers

This package, after catkin building, doesn't generate any executable, so nothing in it can be rosrun. What changes should I make to the CMakeLists.txt? Your package is a library, but any executable referring to it throws an error saying that there is no main.