BehaviorTree/BehaviorTree.CPP #78: Embedded version of tinyXML is conflicting with ROS Kinetic's pluginlib

v-lopez, 2019-03-28:
I guess this is another iteration of #68, #8 and #4.

Since we jumped from 3.0.2 to 3.0.3 I'm getting stack smashes and similar issues when linking in the same application BehaviorTree and ROS PluginLib.

My guess it has something to do with the TinyXML2, but I'm a bit lost.

```
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Stack trace (most recent call last):
#19   Object "", at 0xffffffffffffffff, in 
#18   Object "/home/victor/other_ws/grasping_pipeline_minimal/devel/.private/pal_grasping_pipeline/lib/pal_grasping_pipeline/grasping_pipeline_bt", at 0x65a638, in _start
#17   Source "../csu/libc-start.c", line 291, in __libc_start_main [0x7ff316bea82f]
      > 236:   auto tree = factory.createTreeFromFile(xml_path, blackboard);
        237:   
        238:   // This publisher can cause deadlocks when exiting
        239: //  BT::PublisherZMQ publisher_zmq(tree);
#15   Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/bt_factory.cpp", line 250, in createTreeFromFile [0x7ff31c79ac1e]
        247: {
        248:     XMLParser parser(*this);
        249:     parser.loadFromFile(file_path);
      > 250:     auto tree = parser.instantiateTree(blackboard);
        251:     tree.manifests = this->manifests();
        252:     return tree;
        253: }
#14   Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/xml_parsing.cpp", line 432, in instantiateTree [0x7ff31c7c24b1]
        429:                               root_blackboard,
        430:                               TreeNode::Ptr() );
        431: 
      > 432:     if( output_tree.nodes.size() > 0)
        433:     {
        434:         output_tree.root_node = output_tree.nodes.front().get();
        435:     }
#13   Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/xml_parsing.cpp", line 640, in recursivelyCreateTree [0x7ff31c7c398d]
        637:             {
        638:                 recursiveStep(node, child_element);
        639:             }
      > 640:         }
        641:     };
        642: 
        643:     auto root_element = tree_roots[tree_ID]->FirstChildElement();
#12   Source "/usr/include/c++/5/functional", line 2267, in operator() [0x7ff31c7c6c70]
       2264:     {
       2265:       if (_M_empty())
       2266: 	__throw_bad_function_call();
      >2267:       return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
       2268:     }
       2269: 
       2270: #if __cpp_rtti
#11   Source "/usr/include/c++/5/functional", line 1871, in _M_invoke [0x7ff31c7c45d1]
       1868:       static void
       1869:       _M_invoke(const _Any_data& __functor, _ArgTypes&&... __args)
       1870:       {
      >1871: 	(*_Base::_M_get_pointer(__functor))(
       1872: 	    std::forward<_ArgTypes>(__args)...);
       1873:       }
       1874:     };
#10   Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/xml_parsing.cpp", line 612, in operator() [0x7ff31c7c3584]
        609:     {
        610:         auto node = createNodeFromXML(element, blackboard, parent);
        611:         output_tree.nodes.push_back(node);
      > 612: 
        613:         if( node->type() == NodeType::SUBTREE )
        614:         {
        615:             auto parent_bb = output_tree.blackboard_stack.back();
#9    Source "/home/victor/other_ws/grasping_pipeline_minimal/src/BehaviorTree.CPP/src/xml_parsing.cpp", line 480, in createNodeFromXML [0x7ff31c7c2821]
        477:         {
        478:             const std::string attribute_name = att->Name();
        479:             if (attribute_name != "ID" && attribute_name != "name")
      > 480:             {
        481:                 remapping_parameters[attribute_name] = att->Value();
        482:             }
        483:         }
#8    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7ff3175d4c4b, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)
#7    Source "/usr/include/c++/5/bits/basic_string.tcc", line 216, in _M_construct<char const*> [0x6ac4f1]
        213:       {
        214: 	// NB: Not required, but considered best practice.
        215: 	if (__gnu_cxx::__is_null_pointer(__beg) && __beg != __end)
      > 216: 	  std::__throw_logic_error(__N("basic_string::"
        217: 				       "_M_construct null not valid"));
        218: 
        219: 	size_type __dnew = static_cast<size_type>(std::distance(__beg, __end));
#6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7ff31756914e, in std::__throw_logic_error(char const*)
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7ff317540918, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7ff317540700, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7ff3175406b5, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21", at 0x7ff31754284c, in __gnu_cxx::__verbose_terminate_handler()
#1    Source "/build/glibc-Cl5G7W/glibc-2.23/stdlib/abort.c", line 89, in abort [0x7ff316c01029]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 54, in raise [0x7ff316bff428]
Aborted (Signal sent by tkill() 6468 1000015)
```

Using a very simple tree:
```
<root main_tree_to_execute="BehaviorTree">
    <!-- ////////// -->
    <BehaviorTree ID="BehaviorTree">
        <Sequence>
            <SetBlackboard output_key="grasped" value="false"/>
        </Sequence>
    </BehaviorTree>
</root>
```

facontidavide, 2019-03-28:
Ok, I will get rid of the embedded version  once and for all...

facontidavide, 2019-03-29:
can you please send me a test package to reproduce this issue. I will try to fix it today if I can

facontidavide, 2019-03-29:
I just pushed a commit that MIGHT solve the issue

v-lopez, 2019-03-29:
https://github.com/v-lopez/bt_test_package

With pluginlib it crashes, if you comment it from CMakeLIsts it doesn't crash. I'll test your new version.

v-lopez, 2019-03-29:
As far as I can see, that commit solves the issue! I'll do more digging to make sure that I'm not missing anything but it looks good. Thanks!

closed 2019-03-29 by facontidavide