BehaviorTree/BehaviorTree.CPP #79: Export library definitions

Thordreck, 2019-03-29:
Hi,
I think it would be nice to export some definitions used by the library so external applications can access them. For example, if the library has been compiled with ZMQ support and trees can be monitored.
Something like this could be added at the end of `CMakeLists.txt`:
```
if( ZMQ_FOUND )
    target_compile_definitions(${BEHAVIOR_TREE_LIBRARY} PUBLIC ZMQ_FOUND)
endif()
```

facontidavide, 2019-04-01:
I don't have strong opinions about it...

Thordreck, 2019-04-01:
Well, the main use case I had for this is a generic behavior tree launcher that links against your library. I wanted to be able to #define out the code that instantiates and uses the `BT::PublisherZMQ` logger in case the library was not compiled with `ZMQ`.

facontidavide, 2019-04-02:
done

closed 2019-04-02 by facontidavide