BehaviorTree/BehaviorTree.CPP #80: Infinite loop

BriceRenaudeau, 2019-04-01:
Hi,
I would like to know if there is a simple way to make an infinite loop.
The "**RetryUntilSuccesful**" node has a limit **num_attempts**.

It is possible to make possible by changing **retry_node.cpp** (line 55) :  
while (try_index_ < max_attempts_)
into : 
while (try_index_ < max_attempts_ || max_attempts_==-1)
And set **max_attempts_**  as an int.

Or should I create my own DecoratorNode.

facontidavide, 2019-04-01:
I guess this change makes lot of sense for many users.
I will include it.

facontidavide, 2019-04-03:
Done :)

closed 2019-04-03 by facontidavide

BriceRenaudeau, 2019-04-09:
Thank you it's really good.

You forgot to implement it in the retry_node :
(line 55) while (try_index_ < max_attempts_ || max_attempts_ == -1)

aled96, 2021-12-16:
Hi @facontidavide ,
The aformentioned change was also done for ROS ?

Because I installed from: sudo apt-get install ros-melodic-behaviortree-cpp-v3 but setting -1 as num_cycles allows to make only one iteration.
Should I modify directly the .h file ?

Thanks