BehaviorTree/BehaviorTree.CPP #82: ReactiveSequence

JonasTietz, 2019-04-02:
Hello,
is there a way right now to have a ReactiveSequence, which does not tick the next children of one nodes returns RUNNING. Basically I want the reactive behavior, where the sequence always starts ticking from the beginning (so Sequence and SequenceStart are no option) but I still want the nodes to execute one after another.
Am I missing something or is this a niche usecase, which I schould implement myself?

miccol, 2019-04-03:
Hi Jonas, thank you spotting this.

> Am I missing something or is this a niche usecase, which I schould implement myself?

It not a niche usecase but in fact one of the main features of a reactive sequence. The library used to have (reactive) sequence nodes with the behavior you describe. Maybe something changed with the Version 3? @facontidavide. 

Probably adding a  `haltChildren(index + 1);` and  `return NodeStatus::RUNNING;` after this case may solve the issue.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/26a5fcea97c8b7a8c1e85156c2605c068d0e86b2/src/controls/reactive_sequence.cpp#L30-L33

facontidavide, 2019-04-03:
Hi @miccol 

you are 100% right. That is totally a brain fart from my side. I will fix it

facontidavide, 2019-04-03:
Nevertheless, I still need to find a good solution for the problem mentioned in #71, i.e that Reactive nodes can't have multiple asynchronous children...

miccol, 2019-04-03:
> Nevertheless, I still need to find a good solution for the problem mentioned in #71, i.e that Reactive nodes can't have multiple asynchronous children...

I have an idea about that, but it makes some assumptions on the children. Shall we discuss it on a separated issue?

closed 2019-04-03 by facontidavide

facontidavide, 2019-04-03:
discussion moved to #83

MarcelSoler, 2019-07-24:
Hi!
Something similar is happening to me. I have the following reactive sequence:

[image]

ExPath return running on the first tick and then is never ticked again... Getpath is ticked properlly all the time (it returns succed).

I am using ros melodic and ros-melodic-behaviortree-cpp-v3.