BehaviorTree/BehaviorTree.CPP #87: Fallback/Sequence states inside a Sequence

bwehbe, 2019-04-09:
Hello, I am trying to use a Fallback node inside a sequence. The problem I get is when the Fallback goes into SUCCESS and the next child in the sequence goes to RUNNING, on the next tick the Fallback goes back into RUNNING and reruns all of its children. This is not the case though with an action node, when the action node goes into SUCCESS it will never be reset to RUNNING the next time.

Here is the code I am running. `AlwaysRunning` is simply a node that always return a RUNNING state.

```xml
<root main_tree_to_execute="BehaviorTree">
    <include path="Check.xml" />
    <include path="subtrees/Talk.xml" />
    <BehaviorTree ID="BehaviorTree">
        <Sequence name="Main seq">
            <SayHello />
            <Fallback name="OpenGrip Fallback"> 
                    <Sequence name="Open Grip">
                        <OpenGripper/>
                        <Timeout name="Open Close Timeout" msec="1">
                            <AlwaysRunning/>
                        </Timeout>
                    </Sequence>
                <CloseGripper />
            </Fallback>
            <AlwaysRunning/> 
        </Sequence>
    </BehaviorTree>
</root>
```
The problem is that when the Fallback "OpenGrip Fallback" is finished with a SUCCESS, and the last child of the Sequence, `AlwaysRunning`, switches from IDLE to RUNNING. The Fallback goes back to RUNNING on the next tick.

Is this the expected behavior?

facontidavide, 2019-04-17:
I should double check, but:

- OpenGripper return SUCCESS, therefore Timeout is executed
- Timeout should fail, because the child never return SUCCESS.
- CloseGripper should always be executed

I will write a unit test to check this. Are OpenGripper and CloseGripper synchrinous or ansynchronous actions?

As a consequence, CloseGripper is (or should be) executed.

facontidavide, 2019-04-17:
I implemented a dummy version of your code and I can not observe the behavior you describe.

SayHello, OpenGripper and CloseGripper are executed only once.

This is the log:

```
[1555493711.279]: Main seq                  IDLE -> RUNNING
[1555493711.280]: SayHello                  IDLE -> RUNNING
Hello
[1555493711.280]: SayHello                  RUNNING -> SUCCESS
[1555493711.280]: OpenGrip Fallback         IDLE -> RUNNING
[1555493711.280]: Open Grip                 IDLE -> RUNNING
[1555493711.280]: OpenGripper               IDLE -> RUNNING
Open gripper
[1555493711.280]: OpenGripper               RUNNING -> SUCCESS
[1555493711.280]: Open Close Timeout        IDLE -> RUNNING
[1555493711.280]: AlwaysRunning             IDLE -> RUNNING
[1555493711.281]: AlwaysRunning             RUNNING -> IDLE
[1555493726.199]: Open Close Timeout        RUNNING -> FAILURE
[1555493726.199]: OpenGripper               SUCCESS -> IDLE
[1555493726.199]: Open Close Timeout        FAILURE -> IDLE
[1555493726.199]: Open Grip                 RUNNING -> FAILURE
[1555493726.199]: CloseGripper              IDLE -> RUNNING
Close gripper
[1555493726.199]: CloseGripper              RUNNING -> SUCCESS
[1555493726.199]: Open Grip                 FAILURE -> IDLE
[1555493726.199]: CloseGripper              SUCCESS -> IDLE
[1555493726.199]: OpenGrip Fallback         RUNNING -> SUCCESS
[1555493726.199]: AlwaysRunning             IDLE -> RUNNING
```

You can find the code here: https://gist.github.com/facontidavide/658231b2a97efa2c435012126d107d4a

facontidavide, 2019-05-10:
Do you still have the problem? have you check my dummy version?

bwehbe, 2019-05-15:
Hi, sorry for the late reply. I forgot to mention before that I was on tag 2.4.0. The `AlwaysRunning` node was synchronous, which is apparently not possible anymore on 3.0.6.

So switching to 3.0.6 and using an async node for the `AlwaysRunning` solves the issue, and I get exactly the same results you got.

It is although still worth mentioning that in version 2, this bug is there. Simply defining the `AlwaysRunning` node as:
```cpp
BT::NodeStatus AlwaysRunning()
{
    return BT::NodeStatus::RUNNING;
}
```
and running the xml file I mentioned above, gives the following results
```bash
[1557922784.935]: Main seq                  IDLE -> RUNNING
[1557922784.935]: SayHello                  IDLE -> RUNNING
Robot says: "Hello!!!"
[1557922784.935]: SayHello                  RUNNING -> SUCCESS
[1557922784.935]: Fallback                  IDLE -> RUNNING
[1557922784.935]: Open Grip                 IDLE -> RUNNING
[1557922784.935]: OpenGripper               IDLE -> RUNNING
GripperInterface::open
[1557922784.935]: OpenGripper               RUNNING -> SUCCESS
[1557922784.935]: Open Close Timeout        IDLE -> RUNNING
[1557922784.935]: AlwaysRunning             IDLE -> RUNNING
[1557922784.936]: AlwaysRunning             IDLE -> RUNNING
[1557922784.936]: AlwaysRunning             RUNNING -> IDLE
[1557922784.936]: Open Close Timeout        RUNNING -> FAILURE
[1557922784.936]: OpenGripper               SUCCESS -> IDLE
[1557922784.936]: Open Close Timeout        FAILURE -> IDLE
[1557922784.936]: Open Grip                 RUNNING -> FAILURE
[1557922784.936]: CloseGripper              IDLE -> RUNNING
GripperInterface::close
[1557922784.936]: CloseGripper              RUNNING -> SUCCESS
[1557922784.936]: Open Grip                 FAILURE -> IDLE
[1557922784.936]: CloseGripper              SUCCESS -> IDLE
[1557922784.936]: Fallback                  RUNNING -> SUCCESS
[1557922784.936]: AlwaysRunning             IDLE -> RUNNING
[1557922784.936]: Fallback                  SUCCESS -> RUNNING
[1557922784.936]: Open Grip                 IDLE -> RUNNING
.
.
.
```
The issue was in the step before the last, `[1557922784.936]: Fallback                  SUCCESS -> RUNNING`.

Thanks for the feedback, and sorry again for the late reply :)

closed 2019-05-15 by facontidavide