BehaviorTree/BehaviorTree.CPP #88: Right way to create service client nodes

BriceRenaudeau, 2019-04-11:
Hello,
I am currently using your behaviour tree to send ROS service call and ROS action goal.

I have a bunch of them. I currently make one node per service(action) and pass the topic as a parameter in the builder.
But I can't pass the msg type like this.

So I would like to know if there is a proper way to make a single ClientServeur(Action)Node and then instantiate them with the correct type of message.

maxlein, 2019-07-11:
Hi,  

I am also looking at a nice way to do this.  
I looked in the navigation2 stack and found this [action node](https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp)  

But its done with v2 of this library, as onInit() is outdated from what I read and is also not called in my example.  

Now I have the problem how I could initialize the service client in a base class without passing the node via constructor.  Ports are also not possible ...

MarcelSoler, 2019-07-24:
Hi,
Here you have some examples:
https://github.com/BehaviorTree/BehaviorTree.ROS

facontidavide, 2019-07-28:
Thank you @MarcelSoler . Yes, that example is exactly the one you should follow.

https://github.com/BehaviorTree/BehaviorTree.ROS/tree/master/src/actions

Cheers

closed 2019-07-28 by facontidavide