BehaviorTree/BehaviorTree.CPP #91: Parsing ROS standard msgs

BriceRenaudeau, 2019-04-23:
Hello,
I am using the behavior tree with ROS and I would like to know if will integrate parsing for all the standard messages?

I already did it for geometry_msgs::Pose2D and nav_msgs::Path

```c++
// Template specialization to converts a string to bool.
  template <> inline bool convertFromString(StringView str)
  {
    if( str == "true") return true;
    else return false;
  }

  // Template specialization to converts a string to Pose2D.
  template <> inline geometry_msgs::Pose2D convertFromString(StringView str)
  {
    //printf("Converting string: \"%s\"\n", str.data() );
    // We expect real numbers separated by semicolons
    auto parts = splitString(str, ';');
    if (parts.size() != 3)
    {
      throw RuntimeError("invalid input)");
    }
    else{
      geometry_msgs::Pose2D output;
      output.x     = convertFromString<double>(parts[0]);
      output.y     = convertFromString<double>(parts[1]);
      output.theta = convertFromString<double>(parts[2]);
      return output;
    }
  }

  // Template specialization to converts a string to Path
  template <> inline nav_msgs::Path convertFromString(StringView str)
  {
    //printf("Converting string: \"%s\"\n", str.data() );
    nav_msgs::Path output;
    // We expect real numbers separated by semicolons
    auto poses = splitString(str, ' ');
    for (unsigned int i=0; i< poses.size();i++)
    {
      auto parts = splitString(poses[i], ';');
      if (parts.size() != 7)
      {
        throw RuntimeError("invalid input)");
      }
      else
      {
        geometry_msgs::PoseStamped pose;
        pose.pose.position.x = convertFromString<double>(parts[0]);
        pose.pose.position.y = convertFromString<double>(parts[1]);
        pose.pose.position.z = convertFromString<double>(parts[2]);
        pose.pose.orientation.w = convertFromString<double>(parts[3]);
        pose.pose.orientation.x = convertFromString<double>(parts[4]);
        pose.pose.orientation.y = convertFromString<double>(parts[5]);
        pose.pose.orientation.z = convertFromString<double>(parts[6]);
        output.poses.push_back(pose);
      }
    }
    return output;
  `}
```

facontidavide, 2019-04-24:
> I would like to know if will integrate parsing for all the standard messages?

No, It will not. I can provide some low hanging fruit like __geometry_msgs__, but in general I can not decide for you how to parse any possible message in ROS from string.

If I had this in the future, it will be in this repository (still under development): https://github.com/BehaviorTree/BehaviorTree.ROS

facontidavide, 2019-07-28:
I am closing this issue, because it is clearly a WONTFIX.

Remember that implementing convertFromString<> is optional, not mandatory.
it is required only if you want to parse the message from string.

How would you parse from string something like an image or pointcloud? Why would you want to do something like this.

You can use as type of a DataPort any C++ object, ROS or not, thought.

closed 2019-07-28 by facontidavide