BehaviorTree/BehaviorTree.CPP #97: ROS 2 Dashing Release

nuclearsandwich, 2019-05-16:
@facontidavide it would be great if we could get a release of the BehaviorTree.CPP repository into the upcoming ROS 2 Dashing release in order to unblock the release of navigation2 http://repo.ros2.org/status_page/blocked_releases_dashing.html?q=behaviortree_cpp&s=6&r=1

Thanks!

facontidavide, 2019-05-22:
I am trying to fix this, hopefully soon. Any help is welcome, since I have no clue of what is wrong ....

facontidavide, 2019-05-22:
@nuclearsandwich , I just checked if branch ros2-dev compiles fine on Crystal. Do you know if there is any reason why it might compile in Crystal but not Dashing?

facontidavide, 2019-05-22:
I just realized that I am focusing on the package *behaviortree_cpp_v3* whilst navigation2 still depends on the old *behaviortree_cpp*...
That is the reason why I believe it was OK, but it is not...

nuclearsandwich, 2019-05-22:
The changes since Crystal are listed on https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/#changes-since-the-crystal-release I haven't had time to tuck into the build failures myself but hopefully the above page provides some hints.

dirk-thomas, 2019-05-23:
@facontidavide Please see https://github.com/ros/rosdistro/pull/21254/files#r287098420 and ros/rosdistro#21342 and merge BehaviorTree/BehaviorTree.CPP#99.

dirk-thomas, 2019-05-24:
@facontidavide Please also merge #100.

dirk-thomas, 2019-05-24:
@facontidavide Please make a new patch release from the `ros2-dev` branch and run `bloom` for Dashing. Hopefully with the `dev` build passing the `bin` build will turn over too.

facontidavide, 2019-05-28:
fixed

closed 2019-05-28 by facontidavide

dirk-thomas, 2019-05-28:
> Please make a new patch release from the ros2-dev branch

@facontidavide I was asking for a release of the `ros2-dev` branch which is v3, right? Can you please do that one too?