BehaviorTree/BehaviorTree.CPP #98: [Windows][ROS2] CMake Error at C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41

seanyen, 2019-05-22:
I am building `behaviortree_cpp_v3` from source code for ROS2 on Windows and in the `colcon build`, it failed at this error messages:

Any pointers? Thanks!

```
--- output: behaviortree_cpp_v3
-- The C compiler identification is MSVC 19.16.27030.1
-- The CXX compiler identification is MSVC 19.16.27030.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
--- stderr: behaviortree_cpp_v3
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
CMake Warning at CMakeLists.txt:39 (message):
-- Detecting C compiler ABI info
  ZeroMQ NOT found.  Skipping the build of [PublisherZMQ] and [bt_recorder].
-- Detecting C compiler ABI info - done

-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe

-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
Traceback (most recent call last):
-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done
  File "C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
-- Detecting CXX compile features

-- Detecting CXX compile features - done
    from ament_package.templates import get_environment_hook_template_path
-- Looking for pthread.h

-- Looking for pthread.h - not found
ModuleNotFoundError: No module named 'ament_package'
-- Found Threads: TRUE  

-- Could NOT find ZMQ (missing: ZMQ_LIBRARIES ZMQ_INCLUDE_DIRS) 
CMake Error at C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
CMake Warning at CMakeLists.txt:39 (message):
  execute_process(C:/opt/python37amd64/python.exe
  ZeroMQ NOT found.  Skipping the build of [PublisherZMQ] and [bt_recorder].
  C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py

  C:/colcon_ws/build/behaviortree_cpp_v3/ament_cmake_package_templates/templates.cmake)

  returned error code 1
-- Found PythonInterp: C:/opt/python37amd64/python.exe (found suitable version "3.7.2", minimum required is "3.7") 
Call Stack (most recent call first):
-- Using PYTHON_EXECUTABLE: C:/opt/python37amd64/python.exe
  C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:38 (include)
Traceback (most recent call last):
  C:/opt/ros/dashing/x64/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  File "C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
  C:/opt/ros/dashing/x64/share/ament_cmake/cmake/ament_cmakeConfig.cmake:38 (include)
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
  CMakeLists.txt:51 (find_package)

---
Failed   <<< behaviortree_cpp_v3	[ Exited with code 1 ]
CMake Error at C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(C:/opt/python37amd64/python.exe
  C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  C:/colcon_ws/build/behaviortree_cpp_v3/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  C:/opt/ros/dashing/x64/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:38 (include)
  C:/opt/ros/dashing/x64/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  C:/opt/ros/dashing/x64/share/ament_cmake/cmake/ament_cmakeConfig.cmake:38 (include)
  CMakeLists.txt:51 (find_package)

-- Configuring incomplete, errors occurred!
```

pankhurivanjani, 2019-06-09:
Hello!
Did you solve this issue ?

facontidavide, 2020-03-22:
would you please try again using master? I am positive that it is fixed now

closed 2020-03-22 by facontidavide

alsora, 2020-07-30:
Hi @facontidavide got the same error building master branch.
I don't have ROS on my system, I got the error while actually building a workspace that contains all the ROS core packages + this one

I had to manually do
```
export ROS_VERSION=2
```
to make it work, is that expected?