BehaviorTree/BehaviorTree.CPP #102: Error loading library when using BT_REGISTER_NODES

hlzl, 2019-06-03:
Hi Davide,

I recently tried to switch from registering the nodes statically to using the plugin and registering them dynamically. I basically adopted the way you dynamically register the dummy nodes in your _t01_ example. However when running my implementation, the loader from the SharedLibrary seems to not find a necessary symbol:
```
ERROR loading library [/home/usr/catkin_ws/devel/lib/libmy_nodes.so]: can't find symbol [BT_RegisterNodesFromPlugin]
```
I could not properly figure out how your macro for `BT_REGISTER_NODES` works but it seems like `BT_RegisterNodesFromPlugin` doesn't properly get declared inside `bt_factory.h`.

Any help on how I could get this to work would be appreciated!

facontidavide, 2019-06-03:
because you should not use it in the header file. it must be inside a cpp file.

closed 2019-06-03 by facontidavide

hlzl, 2019-06-04:
> because you should not use it in the header file. it must be inside a cpp file.

I actually use it in the source file. 
By now I figured out that commit 68b7da7 solves this issue as well.

facontidavide, 2019-06-04:
ok... solved then?