BehaviorTree/BehaviorTree.CPP #103: Design: How to pass received data to a node?

maxlein, 2019-06-06:
I am trying and trying but I don't get it to work.  
Think I am missunderstanding something.  

I have a coroutine node, which sends a request out, and should get status information in.  
Injecting the send function to the node is no problem.  
But receiving data is a problem inside the node.  

I have an async callback function which receives my data and then calls `onIncomingMotionStatus`, which itself is a member function of the node.  

But if I write the received data to a member variable, I get a seg fault. As if the node is not writable from another thread... 

```
NodeBuilder MoveNodeBuilder =
              [&](const std::string& name, const NodeConfiguration& config)
{
   auto moveNode = std::make_unique<MoveNode>( name, config,  [&]( const RawMoveRequest &req ){ sendRawMoveRequest(req); } );
   trackingStatusCallback = [&]( const proto::status::TrackingStatus &status ){ moveNode->onIncomingMotionStatus( status ); };
   return moveNode;
};

factory.registerBuilder<MoveNode>( "MoveNode", MoveNodeBuilder);

```
  
Then I tried to write to the blackboard from the outside status callback directly and then read the port from my node. But this is also not working, the port seems to be never set..  

```
client->subscribe( mqtt::getTrackingStatusTopic( shuttleId ), [ & ]( MSG msg ) -> decltype( auto )
      {
         auto status = proto::msgToProto<proto::status::TrackingStatus>(msg);
         tree->blackboard_stack[0]->debugMessage();

         if(status.status() == status.FINISHED)
         {
            log_info("finished! , TrackingStatus received: {}", status.ShortDebugString() );
            tree->rootBlackboard()->set(BT::FINISHED_IO, true);
         }
        
      });
```

I couldn't find any information on this elsewhere in tutorials or in other issues.

maxlein, 2019-06-16:
Maybe asked a bit simpler:  
Why doesn't it work when I pass data from a different thread to a node?  
It looks as if there were two instances of my node, one which receives the data and one which calls the tick function.  

```
void CustomNode::onIncomingData( int data )
{
   configReceived = true; // member variable, tried with atomic_bool and bool
}

BT::NodeStatus CustomNode::tick( )
{
   if( configReceived ) // never set although onIncomingData is called
   {}
}

// I register my node like this
void registerCustomNode( )
   {
      NodeBuilder NodeBuilder =
              [ & ]( const string& name, const NodeConfiguration& config )
              {
                 auto node = make_unique<CustomNode>( name, config );
                 handleData = [&]( int data){ node->onIncomingData( data ); }; // handleData is a functor
                 return node;
              };

      factory.registerBuilder<CustomNode>( "CustomNode", NodeBuilder );
   }
```

facontidavide, 2019-07-28:
Ok, there is something very weird in your code.

You are capturing `node` by **reference** in the lambda `handleData`, but that unique pointer is invalidated when returned...

I would do something like this:

```C++
NodeBuilder NodeBuilder =
              [ & ]( const string& name, const NodeConfiguration& config )
              {
                 auto node = make_unique<CustomNode>( name, config );
                 auto node_ptr = node.get();
                 handleData = [node_ptr]( int data){ node_ptr->onIncomingData( data ); };
                 return node;
              };
```
Does the code segfault, or just behave in a weird way?

I can try to help you, could you send me a proper example that I can compile and debug?

maxlein, 2019-07-29:
No, doesn't segfault, onIncomingData is called and a member variable is set in it.  
But when I read this variable from the tick function, so from another thread, the variable is still on the old value...  
Currently switching to ROS2, so will look for an example.

facontidavide, 2019-07-30:
If you manage to build an example that run on ROS1, that would be great.

Do you have multiple instances of **CustomNode**? If you do, the object **handleData** is overwritten (I have the feeling that this might be your problem).

facontidavide, 2019-09-11:
Any update from your side? I would like to fix this if possible and also to provide a nice example to people with a similar problem as yours.

maxlein, 2019-09-25:
I have an example in ROS2, not using ROS1 ...
Just don't forget to spin your node if you created a new one for your behavior tree nodes ...

```
#ifndef BEHAVIORTREE_CPP_V3_ROS_CALLBACK_NODE_H
#define BEHAVIORTREE_CPP_V3_ROS_CALLBACK_NODE_H

#include "behaviortree_cpp/bt_factory.h"
#include <atomic>
#include <std_msgs/msg/bool.hpp>
#include <rclcpp/node.hpp>

namespace BT
{
   class RosSimpleSwitchNode : public BT::SyncActionNode
   {
   public:

      RosSimpleSwitchNode( const std::string& name, const BT::NodeConfiguration& config, rclcpp::Node::SharedPtr node )
              : BT::SyncActionNode( name, config ), node( node )
      {
         boolSub = node->create_subscription<std_msgs::msg::Bool>( "switch_topic",
                                                                   rclcpp::SystemDefaultsQoS(),
                                                                   [&]( const std_msgs::msg::Bool::SharedPtr msg )
                                                                             {
                                                                                handleSwitch( msg );
                                                                             });
      }

      BT::NodeStatus tick() override
      {
         RCLCPP_INFO( node->get_logger(), "RosSimpleSwitchNode: switch: %d", switchActive.load() );
         return switchActive ? NodeStatus::SUCCESS : NodeStatus::FAILURE;
      };

   private:
      rclcpp::Node::SharedPtr node;
      rclcpp::Subscription<std_msgs::msg::Bool>::SharedPtr boolSub;
      std::atomic_bool switchActive{false};

      void handleSwitch( const std_msgs::msg::Bool::SharedPtr msg )
      {
         if( msg->data )
         {
            RCLCPP_INFO( node->get_logger(), "RosSimpleSwitchNode: handleSwitch: switch is on");
            switchActive = true;
         }
         else
         {
            RCLCPP_INFO( node->get_logger(), "RosSimpleSwitchNode: switch off ..." );
            switchActive = false;
         }
      }

   };
}

#endif //BEHAVIORTREE_CPP_V3_ROS_CALLBACK_NODE_H

```

closed 2020-03-21 by maxlein