BehaviorTree/BehaviorTree.CPP #107: ROS node crashes when using BT::PublisherZMQ with behaviortree-cpp-v3

DLuensch, 2019-07-12:
Hi,

I have the problem when I'm using the BT::PublisherZMQ logger instead of the BT::StdCoutLogger my node crash immediately. The debugger stops at the position where I create the ZMQ logger.

That's the code:

```c++
void rosMainLoop() const
{
  ros::Rate rate(this->mHz);

  BT::BehaviorTreeFactory lBehaviorTreeFactory = this->registerBehavior();
  BT::Tree lTree = lBehaviorTreeFactory.createTreeFromFile(
      ros::package::getPath("mars_agent_logical_agv") + "/behavior/LogicalAgent.xml");

  // BT::PublisherZMQ lZMQLogger(lTree);
  BT::StdCoutLogger lCoutLogger(lTree);

  while (ros::ok())
  {
    ros::spinOnce();
    lTree.root_node->executeTick();
    rate.sleep();
  }

  // Stop the node's resources
  ros::shutdown();
}
```

I'm using Ubuntu 18.04 with ros melodic. I have installed ros-melodic-behaviortree-cpp-v3. The dependencies libzmq3-dev and libdw-dev are installed.

facontidavide, 2019-07-12:
Thanks for reporting, I will check this out

MarcelSoler, 2019-07-24:
I am using ros melodic  ubuntu 18.04 with  ros-melodic-behaviortree-cpp-v3. And it works for me:

```
 BehaviorTreeFactory factory;
  auto tree = factory.createTreeFromFile(
          ros::package::getPath("test") + "/html/behaviour_tree.xml");
    StdCoutLogger logger_cout(tree);
   PublisherZMQ lZMQLogger(tree);
 ros::Rate rate(100); //100hz
    while( ros::ok())
    {
      tree.root_node->executeTick();
      rate.sleep();
    }

    return 0;
}
```

DLuensch, 2019-07-24:
@MarcelSoler Thanks for your example. I can confirm that it is working if you use:
```cpp
 auto lTree = lBehaviorTreeFactory.createTreeFromFile(
      ros::package::getPath("mars_agent_logical_agv") + "/behavior/LogicalAgent.xml");
```
instead of
```cpp
 BT::Tree lTree = lBehaviorTreeFactory.createTreeFromFile(
      ros::package::getPath("mars_agent_logical_agv") + "/behavior/LogicalAgent.xml");
```
If I don't use auto, the ros node crashes.

DLuensch, 2019-07-24:
Ok, I can confirm that it is not a bug. The reason for the crash must be related to my code! I rebuilded my code completely and the crash disappeared.

closed 2019-07-24 by DLuensch