BehaviorTree/BehaviorTree.CPP #110: Compatibility with Python

varunvp, 2019-07-18:
I was redirected to this repository from here https://github.com/miccol/ROS-Behavior-Tree/issues/28
I wanted to know if it can support python ROS nodes too. If it does, are there any examples? Or is it similar to the way it was used in ROS-Behavior-Trees?

facontidavide, 2019-07-18:
When you say "python ROS node" I am not sure to know what you mean.

Are you talking about a custom ActionNode written in python instead of C++ or you mean that your ActionNode (C++) should talk to an external node in Python?

varunvp, 2019-07-19:
Any one, but the former is preferable. I want to minimize the number of things "talking" to each other. That Python ActionNode should have the ability to call rospy, which will make it a ROS/ActionNode hybrid.

varunvp, 2019-07-20:
Thank you for your replies. I assume that your package is in C++ 11. Can't wait to try them out!
Thanks again for the package!

closed 2019-07-20 by varunvp

varunvp, 2019-08-10:
> In other words, you want to mix in the same application C++ and python...
> 
> I actually investigated this a log time ago in this branch: https://github.com/BehaviorTree/BehaviorTree.CPP/tree/python_bind
> 
> Unfortunately, I got stacked and I run out of development time, being the hardest problem to solve the **type conversion** in DataPorts/blackboard between C++ and Python.
> 
> I am sure there might be solutions, using project like [PyBind11](https://github.com/pybind/pybind11) or [Binder](https://github.com/RosettaCommons/binder).
> 
> But for the time being I can not able to work on this.

Except for those 2 features, how lacking is it in functionality with respect to master? Is it part of v3?

dt1729, 2023-03-29:
@facontidavide Can I work on Python bidings of the same ? Is there documentation that I can refer to?