BehaviorTree/BehaviorTree.CPP #113: Reactive sequence with running asynchronous children

MarcelSoler, 2019-07-29:
Hi!
Something similar is happening to me. I have the following reactive sequence:

[image]

ExPath return running on the first tick and then is never ticked again... Getpath is ticked properlly all the time (it returns succed).

I am using ros melodic and ros-melodic-behaviortree-cpp-v3.

_Originally posted by @MarcelSoler in https://github.com/BehaviorTree/BehaviorTree.CPP/issues/82#issuecomment-514565182_

MarcelSoler, 2019-07-29:
Hi! This is my tree:
```
<root main_tree_to_execute="BehaviorTree">
    <!-- ////////// -->
    <BehaviorTree ID="BehaviorTree">
        <Fallback>
            <RetryUntilSuccesful num_attempts="3">
                <Sequence>
                    <Fallback name="Getgoal">
                        <Condition ID="HaveGoal" have_goal="{have_goal}"/>
                        <Action goal="{goal}" ID="NewGoal" have_goal="{have_goal}"/>
                    </Fallback>
                    <Fallback name="Navigate">
                        <ReactiveSequence name="Navigation">
                            <Action cost="{cost}" goal="{goal}" use_start_pose="false" ID="GetPath" new_path="{path}" new_plan_available="{plan_available}"/>
                            <Action ID="ExePath" new_path="{path}" new_plan_available="{plan_available}" have_goal="{have_goal}"/>
                        </ReactiveSequence>
                        <ForceFailure>
                            <Action ID="Recovery" behavior="clear_costmap"/>
                        </ForceFailure>
                    </Fallback>
                </Sequence>
            </RetryUntilSuccesful>
            <SetBlackboard output_key="{have_goal}" name="goalFailed" value="false"/>
        </Fallback>
    </BehaviorTree>
```

- Getpath is a coroaction.
- ExePath is an async action

facontidavide, 2019-07-30:
- I don't understand why this is a **ReactiveSequence**. Why do you want to tick everytime **GetPath**?

- I don't know how **ExePath** is implemented, but keep in mind that unless you call continuously **getInput()**, ports {path} and {path_available} will not be updated, even if it is "Reactive".

- "ExPath return running on the first tick and then is never ticked again". This is exactly how it is supposed to be. ExePath::tick() should be implemented as a function that is ticked only once from the outside, but that can be stopped if halt() is invoked.

Please send to me your implementation of ExePath, otherwise I can hardly understand if what you are experiencing is correct or not.

Alternatively. Write a failing unit test...

MarcelSoler, 2019-07-30:
Ok for what you said, what I am experiencing is correct. Now I have the idea what to do...
I need and internal loop inside the tick() method of ExePath until it succeeds.

> I don't understand why this is a ReactiveSequence. Why do you want to tick everytime GetPath?
> I don't know how ExePath is implemented, but keep in mind that unless you call continuously getInput(), ports {path} and {path_available} will not be updated, even if it is "Reactive".

GetInput will be continuously called.

One last question: can you clarify when halt()  is invoked, I have an idea but I am not really sure of all cases. Is there any info in the documentation. Thanks

closed 2019-08-19 by MarcelSoler