BehaviorTree/BehaviorTree.CPP #116: How should a reactive sequence work when a coroutine node is one of its children?

IgorAlvesSilva, 2019-08-16:
I was testing having a tree composed of a reactive sequence with two children: an AlwaysSuccess first and then a coroutine that returns running until a certain codition is met. Instead of executing the AlwaysSuccess node again each time the tree is ticked only the coroutine node is executed. That's the same behavior of having the same tree but with a sequence control node instead of the reactive sequence.

hlzl, 2019-11-20:
Reactive Sequences only tick conditions again if one of their asnyc or coroutine nodes returns running. As AlwaysSuccess is an action, it's not getting ticked. Have a look at this [example](https://www.behaviortree.dev/tutorial_04_sequence_star/) from the wiki.

As this issue is also quite old, I vote to close it. @facontidavide

facontidavide, 2020-03-19:
I noticed (thank to both of you) that the documentation was obsolete (fixed).

@hlzl I think your comment is not accurate. There should be no difference between a Condition child or an Action Child.

@IgorAlvesSilva what you described is not supposed to happen. Can you give me a way to reproduce this issue?

facontidavide, 2023-06-07:
A `ReactiveSequence` can have only one single child (the last, usually) in RUNNING mode.
What you are trying to do should just be a regular `Sequence`

closed 2023-06-07 by facontidavide