BehaviorTree/BehaviorTree.CPP #117: Does not work with c++17

scgroot, 2019-09-06:
I cannot compile using c++17 because of an error with BT::optional

To reproduce:
main.cc:
```
#include "behaviortree_cpp/behavior_tree.h"
int main() { return 0; }
```
To compile:
`c++ -I/opt/ros/dashing/include -std=gnu++17 main.cc`

Outputs:
> c++ -I/opt/ros/dashing/include -std=gnu++17 main.cc
> In file included from /opt/ros/dashing/include/behaviortree_cpp/control_node.h:18:0,
>                  from /opt/ros/dashing/include/behaviortree_cpp/controls/parallel_node.h:17,
>                  from /opt/ros/dashing/include/behaviortree_cpp/behavior_tree.h:17,
>                  from main.cc:1:
> /opt/ros/dashing/include/behaviortree_cpp/tree_node.h:117:9: error: ‘optional’ in namespace ‘BT’ does not name a template type
>      BT::optional<T> getParam(const std::string& key) const

facontidavide, 2019-09-08:
Ok, I will have a look. Thanks for reporting

facontidavide, 2019-09-11:
You are reporting an issue with V2, that I am not supporting anymore, since 100% of my current focus is V3.

closed 2019-10-23 by facontidavide