BehaviorTree/BehaviorTree.CPP #118: Does not compile with c++17 option on MSVC 2017

NikolayBrezin, 2019-09-13:
behaviortree_cpp_v3\include\behaviortree_cpp\exceptions.h(27): error C2039: 'to_string': is not a member of 'std::basic_string_view<char,std::char_traits<char>>'

Turning on c++14 works well

commit fdfebcb8c27214a7a58abf56bb4d064ad3390baf (HEAD -> master, origin/master, origin/HEAD)

facontidavide, 2019-09-13:
Thanks for reporting

scgroot, 2019-12-13:
It also does not work on gcc, which means the version released with Eloquent is broken when using c++17.

The cause the nonstd::string_view, which does not match the actual released std::string_view. The to_string function in not in the std release. Kinda ironic that by using c++17 features, it breaks for c++17..

Anyway, changing all the `string_view.to_string()` to `nonstd::to_string(sring_view)` seems to fix the problem. You could also do string.data() instead, but string_view is not guaranteed to be null-terminated. The official way I think is just to do std::string(string_view) explicitly, but that won't work with everything pre c++17.

I can make a pull request if you like? Can this then also be fixed on the current published ros-eloquent-behavior-tree-v3 package?

scgroot, 2019-12-13:
I fixed it in the master, but personally I need it in ros2 (eloquent)

facontidavide, 2019-12-13:
ok, I will merge in both

facontidavide, 2020-03-22:
I think it is fixed now (current master)

closed 2020-03-22 by facontidavide