BehaviorTree/BehaviorTree.CPP #120: Error in convertFromString<double>

moritzknaust, 2019-10-11:
I found a strange error in the convertFromString<double> method. If I use the behaviortree in a rqt-gui plugin, the conversion from string to double is wrong. For example "1.23" is converted to 1 instead of 1.23. This only happens if I use a rqt-gui plugin. When I use the behaviortree in a normal ROS node everything works fine. 

I think this error is related to this topic: [https://www.reddit.com/r/cpp/comments/2e68nd/stdstod_is_locale_dependant_but_the_docs_does_not/](url)
Maybe Qt changes the used locale to another value, but I am not sure about this. 

I have fixed it by using std::istringstream for the conversion instead of std::stod like it is explained in the post. My new version of convertFromString in basic_types.cpp is now the following:

```C++
template <>
double convertFromString<double>(StringView str)
{
    std::istringstream iss(str.data());
    iss.imbue(std::locale("C"));
    double temp;
    iss >> temp;
    return temp;
}
```

This error is very confusing and generates really unexpected behaviors, so can you please fix it? But all in all I really like the library!

moritzknaust, 2019-10-14:
I'm not really an expert in C++ development, so I think I cannot help you much at this point. I only found the error and this was the first solution I found...

facontidavide, 2019-10-14:
I will try to figure out a solution. Thanks for reporting the issue

facontidavide, 2019-11-21:
I found a solution that is faster. Fixed.

http://quick-bench.com/DWaXRWnxtxvwIMvZy2DxVPEKJnE

closed 2019-11-21 by facontidavide