BehaviorTree/BehaviorTree.CPP #121: Eloquent Release Plans

crdelsey, 2019-10-22:
@facontidavide What are your plans for making an ROS2 Eloquent release?

Navigation2 is currently discussing the possibility of making an Eloquent release in the next few weeks since OSRF has enabled it in rosdistro, but we need to coordinate with you and make sure we release after you do your release.

By the way, we've migrated Navigation2 to use version 3 of your library. We do not need a release of version 2 anymore.

CC: @stevemacenski, @mkhansen-intel

facontidavide, 2019-10-30:
ok, Will release it today

facontidavide, 2019-10-30:
Hi, 

there is critical issue that needs some action from your side:

it is related to this #125 

Version 2 and 3 of this library can not be installed simultaneously. 

This means that if Navigation2 is using V2, no one in the entire ros community will be able to use V3 in dashing.

I don't care about the opposite (no one being able to use V2 in Eloquent) because V2 must die.

This means that I MUST solve issue #125 first. As a consequence, we need to synchronize and double check the changes I need to do.

crdelsey, 2019-10-30:
There are two viable options I see:

1. We backport the version 3 update to our Dashing release, and you end-of-life V2.
2. You change the V2 release to install in a different include directory, and we update our dashing release to reference files in that alternate directory.

Did you have any other options in mind?

It looks like I broke some behavior when I updated to V3. I'm in the process of figuring out what's wrong at the moment. I can't do option 1 until that problem is solved, but I hope to solve it this week.

facontidavide, 2019-10-30:
I have decided to do a 3rd option: I changes the directory and target names of V3.

therefore the code to be adapted is the Eloquent one.

have a look to this branch https://github.com/BehaviorTree/BehaviorTree.CPP/tree/ros2-devel

facontidavide, 2019-10-30:
> It looks like I broke some behavior when I updated to V3. I'm in the process of figuring out what's wrong at the moment.

This worry me a lot. May I  help somehow?

facontidavide, 2019-10-30:
https://github.com/ros/rosdistro/pull/22852

crdelsey, 2019-10-30:
> I changes the directory and target names of V3

Ok. I'll update to the new names before we do our Eloquent release.

> This worry me a lot. May I help somehow?

Thanks for the offer. I'll let you know.

crdelsey, 2019-11-01:
FYI - I'm running into #119 after pulling the ros2-devel branch and ros2-3.1.0 tags.

closed 2019-11-21 by facontidavide